hadess/iio-sensor-proxy #190: iio-sensor-proxy does not work on dell XPS 9365

denics, 2017-10-08:
Hi and thanks for this tool! I've just acquired a dell XPS 9365, compiled gnome-common and all the other libraries, but iio-sensor-proxy does not work. At least everything compiles with no problems and run, but then this is the output:

> root@denixps:~# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:12553): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/trigger2
** (process:12553): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device2
** (process:12553): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device2 of type accelerometer at IIO Buffer accelerometer
** (process:12553): DEBUG: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.20.auto/iio:device5
** (process:12553): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.20.auto/iio:device5 of type compass at IIO Buffer Compass
** (process:12553): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/trigger2
** (process:12553): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device2/scan_elements/in_accel_y_en
** (process:12553): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device2/scan_elements/in_accel_x_en
** (process:12553): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device2/scan_elements/in_timestamp_en
** (process:12553): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device2/scan_elements/in_accel_z_en
** (process:12553): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Got type for in_timestamp: is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:12553): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Built channel array for in_timestamp: index: 3, is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:12553): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.20.auto/trigger5
** (process:12553): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.20.auto/iio:device5/scan_elements/in_magn_z_en
** (process:12553): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.20.auto/iio:device5/scan_elements/in_magn_y_en
** (process:12553): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.20.auto/iio:device5/scan_elements/in_rot_from_north_magnetic_tilt_comp_en
** (process:12553): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.20.auto/iio:device5/scan_elements/in_magn_x_en
** (process:12553): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:12553): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:12553): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:12553): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:12553): DEBUG: No new data available on 'iio:device2'
** (process:12553): DEBUG: No new data available on 'iio:device2'
** (process:12553): DEBUG: No new data available on 'iio:device2'
** (process:12553): DEBUG: No new data available on 'iio:device2'

and this is my uname -a

> Linux denixps 4.10.0-35-generic #39-Ubuntu SMP Wed Sep 13 07:46:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Is there any way I can help? Thanks!

hadess, 2017-10-08:
Did you read the README and apply the kernel fix? Otherwise your kernel is too old.

denics, 2017-10-08:
Just finished my upgrade to Ubuntu 17.10 and ... you can update your list of tested devices! Thanks!

closed 2017-10-08 by denics

breucker, 2019-08-31:
Hello,
I'm having problems with iio sensors on my dell 9365, mainly the auto-rotation does not work at all.
It worked fine in ubuntu 17.04 when I got the laptop, and at some point it stopped working, I don't remember exactly at which update of version. I'm now on ubuntu 18.04, and tried different kernels (from 4.15 to 5.2), but the sensors don't even show in any of the following test:
- `/sys/bus/iio` does not exist
- `udevadm info --export-db` has nothing resembling to iio
-  `  G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:9501): DEBUG: 21:09:55.398: Could not find any supported sensors
`
I can see the sensor in the list (`Intel Corporation Sunrise Point-LP Integrated Sensor Hub`) but nothing seems to work in gnome.
I just spent the whole day trying to understand the problem but my technical skills are not good enough... If anyone has an idea it would be great ! I hope I'm in the right place to ask...

I will be more than happy to do some tests! 

Thanks,
Philippe

----
Here's my system's summary:
```
# inxi -Fxmz
System:    Host: <filter> Kernel: 5.2.11-050211-generic x86_64 bits: 64 gcc: 9.2.1 Console: tty 1
           Distro: Ubuntu 18.04.3 LTS
Machine:   Device: un-determined System: Dell product: XPS 13 9365 serial: <filter>
           Mobo: Dell model: 0F8YYT v: A00 serial: <filter> UEFI: Dell v: 1.3.1 date: 01/25/2018
Battery    BAT0: charge: 30.2 Wh 100.0% condition: 30.2/46.0 Wh (66%) model: SMP DELL HMPFH62 status: Full
CPU:       Dual core Intel Core i7-7Y75 (-MT-MCP-) arch: Kaby Lake rev.9 cache: 4096 KB
           flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx) bmips: 6432
           clock speeds: max: 3600 MHz 1: 1154 MHz 2: 1162 MHz 3: 1367 MHz 4: 1612 MHz
Memory:    Array-1 capacity: 16 GB devices: 2 EC: None
           Device-1: N/A size: 8 GB speed: 1867 MT/s type: LPDDR3 part: K4EBE304EB-EGCF
           Device-2: N/A size: 8 GB speed: 1867 MT/s type: LPDDR3 part: K4EBE304EB-EGCF
           Device-3: N/A size: N/A speed: N/A type: N/A part: N/A
           Device-4: N/A size: N/A speed: N/A type: N/A part: N/A
Graphics:  Card: Intel Device 591e bus-ID: 00:02.0
           Display Server: X.Org 1.19.6 driver: i915 Resolution: 1920x1080@59.96hz
           OpenGL: renderer: Mesa DRI Intel HD Graphics 615 (Kaby Lake GT2)
           version: 4.6 Mesa 19.3.0-devel (git-2023716 2019-08-31 bionic-oibaf-ppa) Direct Render: Yes
Audio:     Card Intel Sunrise Point-LP HD Audio driver: snd_hda_intel bus-ID: 00:1f.3
           Sound: Advanced Linux Sound Architecture v: k5.2.11-050211-generic
Network:   Card: Intel Wireless 8265 / 8275 driver: iwlwifi bus-ID: 3c:00.0
           IF: wlp60s0 state: up mac: <filter>
Drives:    HDD Total Size: 1024.2GB (53.8% used)
           ID-1: /dev/nvme0n1 model: PM961_NVMe_SAMSUNG_1024GB size: 1024.2GB
Partition: ID-1: / size: 938G used: 514G (58%) fs: ext4 dev: /dev/nvme0n1p2
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 58.0C mobo: N/A
           Fan Speeds (in rpm): cpu: N/A
Info:      Processes: 305 Uptime: 2:27 Memory: 3110.0/15901.4MB Init: systemd runlevel: 5 Gcc sys: 7.4.0
           Client: Shell (bash 4.4.201) inxi: 2.3.56 
```

```
# lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Device 591e (rev 02)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 02)
00:13.0 Non-VGA unclassified device: Intel Corporation Sunrise Point-LP Integrated Sensor Hub (rev 21)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1d.1 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #10 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Device 9d4b (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
3a:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM961/PM961
[... 144 lines cut ...]
acpi_thermal_rel       16384  1 int3400_thermal
intel_pch_thermal      16384  0
mac_hid                16384  0
sch_fq_codel           20480  1
parport_pc             40960  0
ppdev                  24576  0
lp                     20480  0
parport                53248  3 parport_pc,lp,ppdev
ip_tables              32768  2 iptable_filter,iptable_nat
x_tables               40960  5 xt_conntrack,iptable_filter,xt_addrtype,ip_tables,xt_MASQUERADE
autofs4                45056  2
rtsx_pci_sdmmc         28672  0
nvme                   40960  2
psmouse               155648  0
rtsx_pci               65536  2 rtsx_pci_sdmmc,rtsx_pci_ms
nvme_core              98304  4 nvme
intel_ish_ipc          24576  0
intel_ishtp            49152  1 intel_ish_ipc
wmi                    32768  5 intel_wmi_thunderbolt,dell_wmi,wmi_bmof,dell_smbios,dell_wmi_descriptor
i2c_hid                28672  0
hid                   126976  5 i2c_hid,wacom,usbhid,hid_multitouch,hid_generic
video                  49152  3 dell_wmi,dell_laptop,i915
pinctrl_sunrisepoint    28672  0
pinctrl_intel          28672  1 pinctrl_sunrisepoint
```

hadess, 2019-09-04:
@breucker: this isn't a forum, please contact your distribution to figure out which driver is missing.