hadess/iio-sensor-proxy #191: Garbage values on Lenovo Yoga 2 Pro since ebc348a

dos1, 2017-10-10:
Since the commit ebc348abfea241d02861c09ce05ff759d65b77a0, the iio-sensor-proxy completely misbehaves on my Lenovo Yoga 2 Pro, giving negative light sensor values and setting various orientations on even slightest moves of the laptop.

Checking out the revision e69580583b0062ab2a31a37b4fa94d53c9433550 (one above) is enough to fix it, reverting the troublesome commit fixes it as well.

PeterHindes, 2017-10-10:
i believe the reason may be that it uses the gyroscope in the base not the screen, so level is very ambiguous, the values would need to be rotated 90 degrees to fix this along the middle, to fix it switching between upright and upside-down.

dos1, 2017-10-10:
@PeterHindes Nope, check out the linked commit. It's a wrong type interpretation, with the cast (that apparently broke.ARM build) it works fine.

hadess, 2017-10-11:
@dos1 what values do you typically get from monitor-sensor before and after that commit?

dos1, 2017-10-11:
@hadess 
Before: light sensor works in a predictable manner, showing 20-30 lux in my room right now, 3-4 lux when I cover the sensor with my hand; orientation works reliably (correct orientation that changes only when I change it firmly)

After: light sensor shows some garbage like -10, -2.2, -6.1 lux in my room, but 3-4 lux when I cover the sensor with my hand (hmm, signedness issue?); orientation changes to unexpectable values on laptop's slightest moves.

commit 393c1139 (2017-10-11): iio: Use unaligned access helpers to access buffer values

So as to avoid unaligned access warnings on ARM, or unsigned values be
interpreted as signed.

Closes: #191
closed 2017-10-11 by hadess