hadess/iio-sensor-proxy #192: Dell 5537 - Kabylake GT2 - "Problem.." reading scan element information

AndrewSmithWasTaken, 2017-10-14:
On my system, the sensors are present and detected, I am running kernel 4.10.0-37-generic on Ubuntu 16.04 (Unity), but I have no rotation. Monitor-sensor shows nothing. iio-sensor-proxy service is running wit the following output:

● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: enabled)
   Active: active (running) since Sat 2017-10-14 21:32:14 EEST; 17min ago
 Main PID: 957 (iio-sensor-prox)
   CGroup: /system.slice/iio-sensor-proxy.service
           └─957 /usr/sbin/iio-sensor-proxy

Oct 14 21:32:14 laptop systemd[1]: Starting IIO Sensor Proxy service...
Oct 14 21:32:14 laptop iio-sensor-proxy[957]: ** (process:957): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A
Oct 14 21:32:14 laptop iio-sensor-proxy[957]: ** (process:957): WARNING **: Problem reading scan element information: /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-
Oct 14 21:32:14 laptop systemd[1]: Started IIO Sensor Proxy service.
Oct 14 21:41:04 laptop systemd[1]: Started IIO Sensor Proxy service.

Any help greatly appreciated..!

hadess, 2017-10-14:
Which version of iio-sensor-proxy is this? I would also need the debug output as explained in the README.

Note that your kernel might be too old to receive events, but it shouldn't fail trying to read data.

AndrewSmithWasTaken, 2017-10-15:
Thanks for the quick reply. 

I'm running v1.1 as comes with Ubuntu 16.04. I think I know I'm going to be told to update to the latest version so I'll do that and report back.

Current kernel is 4.10.0-37-generic, but I have tried with 4.12.0, which I saw that you said was working for you.

Meanwhile, debug output as root follows:

** (process:10419): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.12.auto/iio:device1
** (process:10419): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.12.auto/iio:device1 of type accelerometer at IIO Buffer accelerometer
** (process:10419): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.21.auto/iio:device0
** (process:10419): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.21.auto/iio:device0 of type compass at IIO Buffer Compass
** (process:10419): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.12.auto/trigger1
** (process:10419): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10419): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

** (process:10419): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.12.auto/iio:device1/in_scale

** (process:10419): WARNING **: Problem reading scan element information: /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.12.auto/iio:device1
** (process:10419): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.21.auto/trigger0
** (process:10419): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10419): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10419): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10419): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10419): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10419): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10419): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10419): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

AndrewSmithWasTaken, 2017-10-15:
Now running 2.2-0ubuntu1, debug output is:

** (process:3765): DEBUG: Could not find any supported sensors

In addition "gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy" now returns:

Error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files

I'm waay out of my comfort zone here, so apologies if I'm ass-backwards. Any help much appreciated.

hadess, 2017-10-15:
> ** (process:10419): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.12.auto/iio:device1/in_scale

Pretty sure that was fixed in version 2.0 with 12c137bcf15e0c6faf25bd71ed1c74754c5ab6b7

> Now running 2.2-0ubuntu1, debug output is:

Not sure how you installed it, but you might need to reboot the system for some of the tagging to be applied to the sensor devices.

AndrewSmithWasTaken, 2017-10-15:
I've tried rebooting after each install, no dice.

Installed from here: https://www.ubuntuupdates.org/package/core/zesty/main/base/iio-sensor-proxy

AndrewSmithWasTaken, 2017-10-15:
Oof, lost my comments, that stings.

Anyway, now running v2.2-0ubuntu1, on kernel  4.13.7-041307-generic and monitor-sensor is working after at least three reboots.

monitor-sensor output:

monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
+++ Accelerometer appeared
    Accelerometer orientation changed: undefined
--- Light sensor disappeared
    Light changed: 0.000000 (lux)
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: right-up
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: bottom-up
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: bottom-up
    Accelerometer orientation changed: normal

I still have some error messages on the debug output and systemctl status has the float warning, but it looks as though it's working. Now to find a script to leverage this for auto-rotation..

closed 2017-10-15 by AndrewSmithWasTaken