hadess/iio-sensor-proxy #194: No roatation on Acer TravelMate

blobgo, 2017-10-16:
The [Acer TravelMate Spin B1](https://us.hardware.info/product/403176/acer-travelmate-spin-b1-tmb118-rn-p3t8/specifications) can be ordered with Linux (Endless OS) preinstalled. I then installed Fedora 26.

So in the preinstalled version turning the display ionto touch-only mode (upside down) did not work, neither. According to the site I've linked, it does however include an accelerometer sensor, so theoretically it should work.

´´'
$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
Fehler: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files
$ dnf info iio-sensor-proxy
Installierte Pakete
Name         : iio-sensor-proxy
Version      : 2.2
Release      : 1.fc26
Arch         : x86_64
Größe        : 109 k
Quelle       : iio-sensor-proxy-2.2-1.fc26.src.rpm
Paketquelle  : @System
Aus Paketque : anaconda
Zusammenfass : IIO accelerometer sensor to input device proxy
URL          : https://github.com/hadess/iio-sensor-proxy
Lizenz       : GPLv3+
Beschreibung : IIO accelerometer sensor to input device proxy.

$ systemctl stop iio-sensor-proxy
$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:8510): DEBUG: Could not find any supported sensors
```

Running `monitor-sensor` basically always just waits for the proxy. I can start the sensor via `systemctl`, but it does not work neither afterwards.

Fedora 26

I can do more tests/run some commands if you need.

blobgo, 2017-10-16:
BTW: As another question/workaround: Can I somehow rotate/flip the display in GNOME/Wayland from the CLI or so, so I can assign a command as a shortcut?

hadess, 2017-10-16:
Which version of the kernel are you using? Did you check the README to see whether your version was new enough?

blobgo, 2017-10-17:
I did. Maybe you did not see my edit in the OP:
> Kernel 4.13 is the most recent in Fedora and installed here, so it should not be affected by the bug you mention in the Readme.

hadess, 2018-10-09:
There's a second bug referenced in the README now, please make sure that your kernel contains this fix. Ask your distributor if in doubt.

hadess, 2018-10-09:
@dsd, which kernel driver does this use, and does iio-sensor-proxy work with this device with the Endless kernel?

blobgo, 2018-10-09:
Who is @dsd? In any case with Linux kernel 4.18.10-200.fc28.x86_64 (now Fedora 28) this issue is still present. And I cannot install Endless OS now, but IIRC it also did not work with it.

dsd, 2018-10-23:
Sorry, we don't have any info available about how to support the accelerometer on this product (assuming there is one here - Acer systems come in many flavours and the site you linked to is not official). I suggest checking if the sensor is available & working under Windows as a next step. If so, Device Manager can probably give some indication what is the hardware compponent behind this.

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/194

closed 2020-01-09 by hadess