hadess/iio-sensor-proxy #195: Last commit breaks orientation reading on SP3

sarangnemo, 2017-10-21:
The [latest commit](https://github.com/hadess/iio-sensor-proxy/commit/393c1139dd7d64f177656eb76043f17c26986d64) breaks orientation reading on my Surface Pro 3. (Light sensor reading is unaffected.)

`monitor-sensor` reports:
    +++ Accelerometer appeared
        Accelerometer orientation changed: undefined

It works properly after reverting to the previous commit

hadess, 2017-10-21:
Could you please attach the debug output as per README?

sarangnemo, 2017-10-21:
Sorry. I attach the output from both the current version and v2.2-3 for reference. In both, the screen starts in the 'normal' horizontal position, and is rotated to right-up.

Looks like the accelerometer values now read the same value for x y and z.

[log-393c113.txt](https://github.com/hadess/iio-sensor-proxy/files/1404289/log-393c113.txt)
[log-v2.2-3.txt](https://github.com/hadess/iio-sensor-proxy/files/1404286/log-v2.2-3.txt)

smithfred, 2017-10-24:
I can confirm broken behaviour with 2.3 on a Surface Pro 3: any small movement results in ~ 2-3 rapid random rotations, with the final rotation bearing no relation to the physical orientation.

Downgrading to 2.2 (on Fedora 27) results in working auto-rotation (with kernel 4.14).

commit 4bf3b38a (2017-10-24): iio: Fix 32-bit unaligned helper

0640bf0 added helpers to fetch unaligned memory, but didn't take
the offset to the data location into account, reading from
the header instead.

Closes: #195
closed 2017-10-24 by hadess