hadess/iio-sensor-proxy #199: Is iio-sensor-proxy the right place to put laptop-tablet mode detection for 2-in-1s?

alesguzik, 2017-11-01:
Hi. I have a ThinkPad X1 Yoga Gen2 laptop which can flip into a tablet mode. It has two accelerometers, and I was able to write a script that reliably detects laptop and tablet modes. I rely on iio-sensor-proxy to do screen rotation, and it looks like the right place to put laptop/tablet detection logic as well. Will it make sense if I will work on a patch implementing that functionality for iio-sensor-proxy?

hadess, 2017-11-02:
I'm pretty certain that there's a better hook to know whether the keyboard is bent at least "this much". IIRC, there was an input event that stopped being sent, but nothing was ever done to have a proper interface available for this in the kernel, so nothing to consume it.

alesguzik, 2017-11-02:
On my device, there's an ACPI event "thinkpad_acpi: unhandled HKEY event 0x60c0" which is sent whenever lid crosses a certain threshold. Unfortunately, unlike on some previous models, it's the same event for both laptop to tablet and tablet to laptop transitions. Also, unlike on other previous models, the event is not sent every second when in tablet mode. Because of this, even if ACPI event would be used to trigger mode check, detection have to rely on accelerometers.

hadess, 2017-11-03:
I think that's a worthwhile route to go down. An ACPI event sent when the mode changes from laptop to tablet is pretty much how the lid switch works itself. Ideally, you would be able to find something else to tell you whether the new state is "tablet" or "laptop" when that event happens (without relying on accelerometer readings), and export that as a "switch" input device, just like the lid switch.

Further integration could happen when the switch is available to user-space.

hadess, 2017-11-06:
For what it's worth, a number of devices could be using this patch:
https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg1528662.html

hadess, 2017-11-06:
libinput already has support for `SW_TABLET_MODE`, which means that if that kernel patch actually works on your machine, you'd just need to make sure that your X11 setup uses libinput, or that you use gnome-shell's Wayland session, and the internal touchpad and keyboards would automatically be disabled when the device is put into tablet mode.

In the end, there should be no need for accelerometers.

closed 2017-11-06 by hadess