hadess/iio-sensor-proxy #200: Lenovo Yoga 910 - AmbientLight and Accelerometer not working

henrythor, 2017-11-14:
Here's debug output from running iio-sensor-proxy.service from boot https://pastebin.com/GUQ9PAGN

Kernel version: 4.13.12-1-ARCH
iio-sensor-proxy version: 2.4-3

Seems there's some hardware quirk that's preventing iio-sensor-proxy from reading data from these sensors.

Any suggestions?

henrythor, 2017-12-19:
Accelerometer is working now but only after a sleep and wake-up. The ambient light sensor isn't outputting anything.

hadess, 2017-12-19:
Check the README for the exact versions that are broken. Bear in mind that it's broken again since an unknown version:
https://www.spinics.net/lists/linux-iio/msg36003.html

RussianNeuroMancer, 2019-01-14:
Do you still have this issue with latest kernel and iio-sensor-proxy?

hadess, 2019-05-13:
I'm going to assume this works correctly now. Please reopen if not.

closed 2019-05-13 by hadess