hadess/iio-sensor-proxy #201: Local quirks not registering

Split7fire, 2017-11-24:
I have Chuwi Hibook 2-in-1 tablet pc with Cherrytrail Intel processor. On Fedora 27 accelerometer (Bosch BOSC0200) works fine, except it is rotated 90 deg counterclockwise. I managed to read README and get modalias and pn. (Udevadm log attached) After making changes to 61-sensor-local.hwdb, I issued systemd-hwdb update, then systemctl restart iio-sensor-proxy, but no visible differences to rotation. I also tried reboot and no luck.

P.S. I know about similar issue https://github.com/hadess/iio-sensor-proxy/issues/169, that's not my case, because i'm using Fedora 27 with systemd 234.

[61-sensor-local.hwdb.txt](https://github.com/hadess/iio-sensor-proxy/files/1501898/61-sensor-local.hwdb.txt)
[udevadm.log](https://github.com/hadess/iio-sensor-proxy/files/1501900/udevadm.log)

hadess, 2017-11-24:
> E: ACCEL_MOUNT_MATRIX=1, 0, 0; 0, 1, 0; 0, 1, 0

This mount matrix is broken. You have "y" appearing twice in the matrix. You'll also want to attach the debug output of iio-sensor-proxy as per the README, and verify you have the latest version of iio-sensor-proxy.

Split7fire, 2017-11-24:
Okay, that particular hwdb file might be wrong, but I tried 1,0,0; 0,1,0; 0,0,1 and nothing. Also I cannot find detailed discription of this matrix.

Split7fire, 2017-11-25:
IIO-sensor-proxy package in Fedora 27 has 2.4-1.fc27 version.

Split7fire, 2017-11-25:
Feeling lucky today, I tried to set ACCEL matrix to similar Chuwi hardware (https://github.com/systemd/systemd/blob/master/hwdb/60-sensor.hwdb) ACCEL_MOUNT_MATRIX=0, -1, 0; -1, 0, 0; 0, 0, 1 , and it works as expeted - bottom is bottom, up is up.
One more thing: while changing 61-sensor-local.hwdb and systemd-hwdb update and udevadm trigger -y `dirname $(udevadm info -n "/dev/iio:deviceXXX" -q path)` and systemctl restart iio-sensor-proxy can't reload ACCEL matrix. But if I make an error in this matrix, then systemd-hwdb update gave an error, that bad rule is in 61-sensor-local.hwdb.

Also I want to PR this matrix to systemd, can I reference this issue to PR?

hadess, 2017-11-26:
I'm sorry, I don't understand what you tested and what "can't reload ACCEL matrix". In any case, this isn't an iio-sensor-proxy problem.

You can certainly reference this issue, though I'm not sure what information is contained here that would help with the PR.

closed 2017-11-26 by hadess

Split7fire, 2017-11-26:
@hadess I mean that ACCEL matrix stays the same as it was when PC was booted. And no matter if I issued:
`systemd-hwdb update` then `udevadm trigger -y dirname $(udevadm info -n "/dev/iio:device0" -q path)` and `systemctl restart iio-sensor-proxy `
ACCEL matrix dont change. But then you just reboot, matrix change.

hadess, 2017-11-27:
Might be the hwdb udev builtin function not importing the new value on top of the other one. If you can easily reproduce the problem, please file this as a bug against systemd as well, and we'll either make changes to udev, or update the documentation to take this into account.