hadess/iio-sensor-proxy #204: Dell Inspiron 15 accelerometer issue

trjw, 2017-12-17:
Hi,

My issue has many common elements from previous ones on this topic, but none of the recommended fixes (most relevant - a modern kernel) have worked so I'm opening a new issue.

` jwilliams@jwilliams-Inspiron-15-7579:~$ cat /etc/issue
Ubuntu 16.04.3 LTS \n \l`

` jwilliams@jwilliams-Inspiron-15-7579:~$ uname -a
Linux jwilliams-Inspiron-15-7579 4.14.6-041406-generic #201712140930 SMP Thu Dec 14 09:31:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

```
root@jwilliams-Inspiron-15-7579:~/src/systemd# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:16919): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.11.auto/iio:device3
** (process:16919): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.11.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:16919): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.16.auto/iio:device0
** (process:16919): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.16.auto/iio:device0 of type compass at IIO Buffer Compass
** (process:16919): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.11.auto/trigger3
** (process:16919): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:16919): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

** (process:16919): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.11.auto/iio:device3/in_scale

** (process:16919): WARNING **: Problem reading scan element information: /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.11.auto/iio:device3
** (process:16919): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.16.auto/trigger1
** (process:16919): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:16919): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:16919): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:16919): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:16919): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:16919): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:16919): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:16919): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
```

```
jwilliams@jwilliams-Inspiron-15-7579:~$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = false;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};

```

The relevant sysfs files exist and respond as expected when I move the laptop around:
```
jwilliams@jwilliams-Inspiron-15-7579:/sys/bus/iio/devices/iio:device3$ ls
buffer                   in_accel_offset              in_accel_y_raw  scan_elements
current_timestamp_clock  in_accel_sampling_frequency  in_accel_z_raw  subsystem
dev                      in_accel_scale               name            trigger
in_accel_hysteresis      in_accel_x_raw               power           uevent

jwilliams@jwilliams-Inspiron-15-7579:/sys/bus/iio/devices/iio:device3$ while [ 1 ]; do cat in_accel_z_raw; done 
-427358
-447918
-447918
-437656
-437656
-437656
-424864
-424864
-424864
-424864
-429190

```

However monitor-sensors does nothing at all
```
jwilliams@jwilliams-Inspiron-15-7579:/sys/bus/iio/devices/iio:device3$ monitor-sensor 
<crickets chirping, tumbleweeds blowing past...>
```

I have seen the outputs of the in_accel_*_raw files lock up, and doing a suspend resume cycle got them moving again, however nothing I do gets monitor-sensors to connect to iio-sensor-proxy.

I'm using the iio-sensor-proxy package from Ubuntu 16.04 repository.

I note, the same as others have previously, that the iio-sensor-proxy process looks for a missing `in_scale` value, whereas the sysfs file is actually called `in_accel_scale`.  

the gdbus output indicates no acceleraometer is found, when that's clearly not the case.

Any suggestions would be much appreciated.

Thanks,

John

hadess, 2017-12-17:
Looks like Ubuntu's iio-sensor-proxy version in 16.04 is 7 releases behind the latest upstream. You're better off asking Ubuntu for help, I can't support this old a release.

trjw, 2017-12-18:
OK thanks.  I looked into building the latest from source but quickly got into library version dependency hell and aborted.

closed 2017-12-18 by trjw