hadess/iio-sensor-proxy #205: DE-agnostic screen rotation?

adrfantini, 2017-12-17:
I have iio-sensors-proxy set up correctly and `monitor-sensor` reports my laptop, an Acer Spin 5, as correctly rotating. However, no screen rotation is applied, and I suspect this is because iio-sensor-proxy is not DE-agnostic and requires GNOME, which I do not run. Am I right, or am I missing something? Could iio-sensor-proxy call xrandr directly when under X and not running GNOME?

hadess, 2017-12-17:
iio-sensor-proxy doesn't integrate in GNOME, GNOME uses iio-sensor-proxy.

As mentioned in #101 and #95 and probably others:
There's nothing GNOME specific implemented here, it's a D-Bus interface. The GNOME part is implemented in GNOME, just the same way that the XFCE part needs to be implemented in XFCE.

Maybe the thing in #129 would help you.

closed 2017-12-17 by hadess

adrfantini, 2017-12-17:
#129 is extremely helpful. Thanks. Sorry for not understanding that this was not bound to GNOME by any means.

hadess, 2017-12-17:
That's fine, I should probably add it as a FAQ in the README :)

adrfantini, 2017-12-17:
Anyway, I'll open a feature request in Plasma to integrate this. It works perfectly on a late 2017 Acer Spin 5 by the way.