hadess/iio-sensor-proxy #208: Lenovo Yoga 2 11 - no output from geoclue - Linux Mint 18.3

ross-ops, 2017-12-30:
Hello! I am battling to get the screen of my Lenovo 2 working in tablet mode (hopefully including auto-rotation, touchpad disabled etc.) Only my 'sitting on the sofa with a beer-book/lab book', but even so, I'd like to fiddle with it if I can. I've read through the readme and get to this part `su -s /bin/sh geoclue -c monitor-sensor` which outputs nothing.

Interestingly, the live boot of the latest version of Fedora (26) does interact with the screen rotation sensor, albeit incorrectly (inverts screen 180).

So! The important dumps:
` # G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-prox`
```** (process:20301): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200041.7.auto/iio:device1
** (process:20301): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200041.7.auto/iio:device1 of type ambient light sensor at IIO Buffer Light sensor
** (process:20301): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200073.2.auto/iio:device4
** (process:20301): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200073.2.auto/iio:device4 of type accelerometer at IIO Buffer accelerometer
** (process:20301): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200083.4.auto/iio:device0
** (process:20301): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200083.4.auto/iio:device0 of type compass at IIO Buffer Compass
** (process:20301): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200073.2.auto/trigger4
** (process:20301): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:20301): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0

** (process:20301): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200073.2.auto/iio:device4/in_scale

** (process:20301): WARNING **: Problem reading scan element information: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200073.2.auto/iio:device4
** (process:20301): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200041.7.auto/trigger1
** (process:20301): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:20301): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:20301): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8386.0003/HID-SENSOR-200083.4.auto/trigger0
** (process:20301): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:20301): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:20301): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:20301): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:20301): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:20301): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:20301): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:20301): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:20301): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
```
My system dump:
`inxi -F`
```
System:    Host: ross-Lenovo-Yoga-2-11 Kernel: 4.10.0-42-generic x86_64 (64 bit)
           Desktop: Cinnamon 3.6.7  Distro: Linux Mint 18.3 Sylvia
Machine:   System: LENOVO (portable) product: 20332 v: Lenovo Yoga 2 11
           Mobo: LENOVO model: AIUU1 v: 31900042STD Bios: LENOVO v: 92CN23WW(V1.03) date: 11/26/2013
CPU:       Quad core Intel Pentium N3520 (-MCP-) cache: 1024 KB 
           clock speeds: max: 2415 MHz 1: 803 MHz 2: 1193 MHz 3: 826 MHz 4: 759 MHz
Graphics:  Card: Intel Atom Processor Z36xxx/Z37xxx Series Graphics & Display
           Display Server: X.Org 1.18.4 driver: intel Resolution: 1366x768@60.02hz
           GLX Renderer: Mesa DRI Intel Bay Trail GLX Version: 3.0 Mesa 17.0.7
Audio:     Card Intel Atom Processor Z36xxx/Z37xxx Series High Definition Audio Controller
           driver: snd_hda_intel
           Sound: Advanced Linux Sound Architecture v: k4.10.0-42-generic
Network:   Card-1: Broadcom BCM43142 802.11b/g/n driver: wl
           IF: wlp1s0 state: up mac: 80:56:f2:f7:52:bb
           Card-2: Atmel
           IF: null-if-id state: N/A speed: N/A duplex: N/A mac: N/A
Drives:    HDD Total Size: 480.1GB (8.0% used) ID-1: /dev/sda model: SanDisk_Ultra_II size: 480.1GB
Partition: ID-1: / size: 406G used: 33G (9%) fs: ext4 dev: /dev/sda2
           ID-2: swap-1 size: 4.17GB used: 0.01GB (0%) fs: swap dev: /dev/sda3
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 44.0C mobo: N/A
           Fan Speeds (in rpm): cpu: N/A
Info:      Processes: 224 Uptime: 1 day Memory: 1820.8/3832.8MB Client: Shell (zsh) inxi: 2.2.35 
```

I think that covers everything any useful people may wish to know, sorry if I've missed something glaringly obvious. Thanks in advance for the help :)
R

ross-ops, 2018-01-02:
I have realised that this issue may be down to 18.3's repo version of iio-sensor-proxy being old (1.4), so have tried compiling the new version from scratch. 

After a bit of dependency hell, I got version 2 to compile (with warnings), but there still appears not to be any sensor data coming through to geoclue. 

Cheers

R

hadess, 2019-05-13:
There's a warning in the output, which should have better debug nowadays. Try it again and see whether you get a better error message including an explanation as to why it can't read the `in_scale` file.

hadess, 2019-05-14:
Version 2.6 was released, which is what I would advise to use.

hadess, 2020-01-09:
Moveed to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/208

closed 2020-01-09 by hadess