hadess/iio-sensor-proxy #211: How does one go about adding support for new sensors? 2017 Spectre x360 13 - i7 8550u

ilikenwf, 2018-01-26:
Do I need to patch the kernel, or is it just modifying iio-sensor-proxy?

Output from udev:
`P: /devices/LNXSYSTM:00
E: DEVPATH=/devices/LNXSYSTM:00
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXSYSTM:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13885426

P: /devices/LNXSYSTM:00/ACPI0008:00
E: DEVPATH=/devices/LNXSYSTM:00/ACPI0008:00
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:ACPI0008:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13888353

P: /devices/LNXSYSTM:00/INT339B:00
E: DEVPATH=/devices/LNXSYSTM:00/INT339B:00
E: ID_VENDOR_FROM_DATABASE=Interphase Corporation
E: MODALIAS=acpi:INT339B:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13886510

P: /devices/LNXSYSTM:00/INT33D0:00
E: DEVPATH=/devices/LNXSYSTM:00/INT33D0:00
E: ID_VENDOR_FROM_DATABASE=Interphase Corporation
E: MODALIAS=acpi:INT33D0:PNP0C02:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13886465

P: /devices/LNXSYSTM:00/INT3420:01
E: DEVPATH=/devices/LNXSYSTM:00/INT3420:01
E: ID_VENDOR_FROM_DATABASE=Interphase Corporation
E: MODALIAS=acpi:INT3420:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13886510

P: /devices/LNXSYSTM:00/LNXCPU:00
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:00
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13885983

P: /devices/LNXSYSTM:00/LNXCPU:01
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:01
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13885981

P: /devices/LNXSYSTM:00/LNXCPU:02
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:02
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13886131

P: /devices/LNXSYSTM:00/LNXCPU:03
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:03
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=13886148

P: /devices/LNXSYSTM:00/LNXCPU:04
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:04
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXCPU:
[... 5895 lines cut ...]
E: DEVLINKS=/dev/disk/by-id/dm-name-root /dev/disk/by-uuid/5f856ec8-250b-4065-a835-17b29fcdaa49 /dev/disk/by-id/dm-uuid-CRYPT-LUKS1-d87017d56e294c3796096f8cd99a26e9-root /dev/disk/by-label/root_c /dev/mapper/root /dev/gpt-auto-root
E: DEVNAME=/dev/dm-0
E: DEVPATH=/devices/virtual/block/dm-0
E: DEVTYPE=disk
E: DM_ACTIVATION=1
E: DM_NAME=root
E: DM_SUSPENDED=0
E: DM_UDEV_DISABLE_LIBRARY_FALLBACK_FLAG=1
E: DM_UDEV_PRIMARY_SOURCE_FLAG=1
E: DM_UDEV_RULES_VSN=2
E: DM_UUID=CRYPT-LUKS1-d87017d56e294c3796096f8cd99a26e9-root
E: ID_FS_LABEL=root_c
E: ID_FS_LABEL_ENC=root_c
E: ID_FS_TYPE=ext4
E: ID_FS_USAGE=filesystem
E: ID_FS_UUID=5f856ec8-250b-4065-a835-17b29fcdaa49
E: ID_FS_UUID_ENC=5f856ec8-250b-4065-a835-17b29fcdaa49
E: ID_FS_VERSION=1.0
E: MAJOR=254
E: MINOR=0
E: SUBSYSTEM=block
E: TAGS=:systemd:
E: USEC_INITIALIZED=13520816

`

hadess, 2018-01-26:
Unless it's a new _type_ of sensor, only making sure the kernel supports it should be needed. We support drivers written for the IIO, input or hwmon classes of drivers, depending on the type of sensor as well.

hadess, 2018-01-28:
I looked at the original udevadm listing, and it could be that your kernel doesn't have the `hid-sensor-hub` driver available. Might be a good idea to try with a more featureful distro's kernel, such as an Ubuntu or Fedora live CD.

If it still fails, please get in touch with the linux-iio mailing-list, they might be able to help you out.

closed 2018-01-28 by hadess