hadess/iio-sensor-proxy #212: Rotation not working, HP x2 (10-n140nw)

kmbt, 2018-02-03:
> Distro: Debian testing (Buster)
> iio-sensor-proxy updated to 2.4-2 (from Debian Sid package)
> custom kernel, 4.13, all Accelerometer drivers enabled

Rotation not working, `monitor-sensor` doesn't output any data.

```
$ monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
```
### Output of  `$ G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy` 

```
** (process:3164): DEBUG: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200041.1.auto/iio:device4
** (process:3164): DEBUG: Found device /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200041.1.auto/iio:device4 of type ambient light sensor at IIO Buffer Light sensor
** (process:3164): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.11.auto/trigger3
** (process:3164): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.11.auto/iio:device3
** (process:3164): DEBUG: Found device /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.11.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:3164): DEBUG: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.26.auto/iio:device2
** (process:3164): DEBUG: Found device /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.26.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:3164): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.11.auto/trigger3
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.11.auto/iio:device3/scan_elements/in_accel_y_en
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.11.auto/iio:device3/scan_elements/in_accel_x_en
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.11.auto/iio:device3/scan_elements/in_timestamp_en
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.11.auto/iio:device3/scan_elements/in_accel_z_en
** (process:3164): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Got type for in_timestamp: is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:3164): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_timestamp: index: 3, is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:3164): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200041.1.auto/trigger4
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200041.1.auto/iio:device4/scan_elements/in_illuminance_en
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200041.1.auto/iio:device4/scan_elements/in_intensity_both_en
** (process:3164): DEBUG: Got type for in_illuminance: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_illuminance: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.26.auto/trigger2
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.26.auto/iio:device2/scan_elements/in_magn_z_en
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.26.auto/iio:device2/scan_elements/in_magn_y_en
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.26.auto/iio:device2/scan_elements/in_rot_from_north_magnetic_tilt_comp_en
** (process:3164): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.26.auto/iio:device2/scan_elements/in_magn_x_en
** (process:3164): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3164): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3164): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3164): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3164): DEBUG: No new data available on 'iio:device4'
** (process:3164): DEBUG: No new data available on 'iio:device3'
** (process:3164): DEBUG: No new data available on 'iio:device4'
** (process:3164): DEBUG: No new data available on 'iio:device3'
** (process:3164): DEBUG: No new data available on 'iio:device4'
** (process:3164): DEBUG: No new data available on 'iio:device3'
** (process:3164): DEBUG: No new data available on 'iio:device4'
** (process:3164): DEBUG: No new data available on 'iio:device3'
** (process:3164): DEBUG: No new data available on 'iio:device4'
** (process:3164): DEBUG: No new data available on 'iio:device3'
** (process:3164): DEBUG: No new data available on 'iio:device4'
[... 42 lines cut ...]
** (process:3164): DEBUG: No new data available on 'iio:device4'
** (process:3164): DEBUG: No new data available on 'iio:device3'
[... cut ...]
** (process:3164): DEBUG: No new data available on 'iio:device4'
** (process:3164): DEBUG: No new data available on 'iio:device3'
** (process:3164): DEBUG: No new data available on 'iio:device4'
** (process:3164): DEBUG: No new data available on 'iio:device3'
** (process:3164): DEBUG: No new data available on 'iio:device4'
```
[dmesg.txt](https://github.com/systemd/systemd/files/1692051/dmesg.txt)

[kernel_config.txt](https://github.com/systemd/systemd/files/1692050/kernel_config.txt)

[lspci.txt](https://github.com/systemd/systemd/files/1692053/lspci.txt)

kmbt, 2018-02-03:
Polling with:
```
watch -n 0.1 cat /sys/bus/iio/devices/iio\:device*/*raw*
```
actually  makes rotation work as in #180 . The commit mentioned in the README:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=f1664eaacec31035450132c46ed2915fd2b2049a
is applied to my kernel.

kmbt, 2018-02-05:
I've also tested a recent kernel: 4.15.0, with `iio-sensor-proxy` 2.2-1 as well as 2.4-2. The results are the same. Rotation works only with the `cat` trick.

hadess, 2018-02-06:
Kernel broke again:
https://marc.info/?l=linux-iio&m=151136436026917&w=2

marko-lazic, 2018-03-31:
Is there a workaround?

hadess, 2018-03-31:
Finding a working kernel. And if you find the last working kernel, then please tell me, as I can use that when I finally get the time to (try to) root cause this problem again.

micheller, 2018-05-13:
4.15.18-1-MANJARO worked for me, the sensors are working with no `watch ...` stuff needed. But it looks like sleep or hybernation breaks the things, an sensors are no more updated. The `watch` call does not help too.

micheller, 2018-05-21:
Linux Sun2 4.15.18-2-MANJARO got my yoga 920 sensors working even after laptop sleep.

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/212

closed 2020-01-09 by hadess