hadess/iio-sensor-proxy #213: Lenovo Yoga 720 Ambient Light and Accelerometer not working

FlexW, 2018-02-03:
No output for ambient light sensor and accelerometer sensor.
Heres the ouput
```
** (process:17582): DEBUG: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200041.3.auto/iio:device0
** (process:17582): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200041.3.auto/iio:device0 of type ambient light sensor at IIO Buffer Light sensor
** (process:17582): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/trigger1
** (process:17582): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/iio:device1
** (process:17582): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/iio:device1 of type accelerometer at IIO Buffer accelerometer
** (process:17582): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/trigger1
** (process:17582): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/iio:device1/scan_elements/in_accel_y_en
** (process:17582): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/iio:device1/scan_elements/in_accel_x_en
** (process:17582): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/iio:device1/scan_elements/in_timestamp_en
** (process:17582): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/iio:device1/scan_elements/in_accel_z_en
** (process:17582): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 8, shift: 0, mask: 0x255, be: 0
** (process:17582): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 8, shift: 0, mask: 0x255, be: 0

** (process:17582): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/iio:device1/in_scale

** (process:17582): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200073.1.auto/iio:device1/in_offset
** (process:17582): DEBUG: Got type for in_timestamp: is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:17582): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 8, shift: 0, mask: 0x255, be: 0
** (process:17582): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 8, shift: 0, mask: 0x255, be: 0
** (process:17582): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 8, shift: 0, mask: 0x255, be: 0
** (process:17582): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 8, shift: 0, mask: 0x255, be: 0
** (process:17582): DEBUG: Built channel array for in_timestamp: index: 3, is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:17582): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200041.3.auto/trigger0
** (process:17582): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200041.3.auto/iio:device0/scan_elements/in_illuminance_en
** (process:17582): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-7/i2c-0416C300:00/0018:0416:C300.0003/HID-SENSOR-200041.3.auto/iio:device0/scan_elements/in_intensity_both_en
** (process:17582): DEBUG: Got type for in_illuminance: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17582): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17582): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17582): DEBUG: Built channel array for in_illuminance: index: 1, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17582): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:17582): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:17582): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:17582): DEBUG: No new data available on 'iio:device0'
** (process:17582): DEBUG: No new data available on 'iio:device1'
** (process:17582): DEBUG: No new data available on 'iio:device0'
** (process:17582): DEBUG: No new data available on 'iio:device1'
** (process:17582): DEBUG: No new data available on 'iio:device0'
** (process:17582): DEBUG: No new data available on 'iio:device1'
** (process:17582): DEBUG: No new data available on 'iio:device0'
** (process:17582): DEBUG: No new data available on 'iio:device1'
** (process:17582): DEBUG: No new data available on 'iio:device0'
** (process:17582): DEBUG: No new data available on 'iio:device1'
** (process:17582): DEBUG: No new data available on 'iio:device0'
** (process:17582): DEBUG: No new data available on 'iio:device1'
** (process:17582): DEBUG: No new data available on 'iio:device0'
** (process:17582): DEBUG: No new data available on 'iio:device1'
** (process:17582): DEBUG: No new data available on 'iio:device0'
** (process:17582): DEBUG: No new data available on 'iio:device1'
```

However if I read out the sensors manually with `cat` everything works fine.

Kernel: 4.13

hadess, 2018-02-06:
Kernel broke again:
https://marc.info/?l=linux-iio&m=151136436026917&w=2

hadess, 2018-05-14:
Also, the version of iio-sensor-proxy is outdated.

hadess, 2019-05-14:
Please let us know whether the problem still happens with version 2.6.

FlexW, 2019-05-14:
I don't know which exact version I have installed (I don't know how to check) but I can confirm that it is now working.

closed 2019-05-14 by FlexW