hadess/iio-sensor-proxy #214: Asus UX430 light sensor don't report the good values

yodatak, 2018-02-04:
Sorry if it's not the good place to make the report , but with running i got the same reporting of 400 lux even when i hide the sensor , and it do have in gnome the automatic lightning switch but i think its a driver issue or something ?

So there is the debuging : 
```
G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:12646): DEBUG: Found IIO poll light at /sys/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00/iio:device0
** (process:12646): DEBUG: Found device /sys/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00/iio:device0 of type ambient light sensor at IIO Polling Light sensor
** (process:12646): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:12646): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:12646): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:12646): DEBUG: Light level sent by driver (quirk applied): 400.000000 (unit: lux)
** (process:12646): DEBUG: Emitted light changed: from 0.000000 to 400.000000
** (process:12646): DEBUG: Light level sent by driver (quirk applied): 400.000000 (unit: lux)
** (process:12646): DEBUG: Light level sent by driver (quirk applied): 400.000000 (unit: lux)
** (process:12646): DEBUG: Light level sent by driver (quirk applied): 400.000000 (unit: lux)
** (process:12646): DEBUG: Light level sent by driver (quirk applied): 400.000000 (unit: lux)
** (process:12646): DEBUG: Light level sent by driver (quirk applied): 400.000000 (unit: lux)

```

Or monitor-sensor i got 

```
monitor-sensor                           
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== No accelerometer
=== Has ambient light sensor (value: 400.000000, unit: lux)
```

with this i got 
```
gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = false;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 400.0;
  };
  node Compass {
  };
};

```
It seem my light sensor is /sys/bus/acpi/devices/ACPI0008:00
and i think its this can't help but not sure https://github.com/danieleds/Asus-Zenbook-Ambient-Light-Sensor-Controller

Thanks for the helping !

hadess, 2018-02-06:
If the value reported by iio-sensor-proxy matches the value reported by the driver, then iio-sensor-proxy is working correctly. You'll need to report the bug to the driver maintainer, either directly or through your distribution.

closed 2018-02-06 by hadess

yodatak, 2018-02-07:
Thanks for helping , its a driver issue so i opend a issue on kernel here https://bugzilla.kernel.org/show_bug.cgi?id=198709

Just place this for tracking