hadess/iio-sensor-proxy #215: Support for HiDPI screens

ManuGithubSteam, 2018-02-10:
Hi

Im useing Gnome 3.26 and the iio works as intendet in normal scaleing (1) or bigger mode (2).

However as 2 is to big and one is to small. I use xrandr to scale the diplay down to 1.25 from 2. This works really well in landscape mode. But when i turn the display the orientation gets switch but not the scaleing. So its all bigger in porträt mode.

Would it be possible with not much hassle to include a scaleing config parmater so it would scale to the correct size when flipping the screen ?

Thanks

hadess, 2018-02-10:
From the README:
> Note that nothing in iio-sensor-proxy is GNOME specific, or relies on GNOME. GNOME and geoclue use the data provided by iio-sensor-proxy, other desktop environments are more than welcome to use this as a basis for their own integration.

iio-sensor-proxy doesn't touch your display configuration, or anything else in your session, it just reads sensors.

closed 2018-02-10 by hadess

ManuGithubSteam, 2018-02-11:
This can be useful in other DEs like XFCE Mate etc, its not a Gnome problem so to speak. 

But i think i see what you mean. However this leaves me at a loss and means i problay need to do iio functions via script that do what i want but are ugly. :-(

Thanks anyway for looking into my direction.

hadess, 2018-02-11:
iio-sensor-proxy doesn't touch the display at all, I can't fix the bug (if it is a bug) mentioned in your comment, even if I wanted to.