hadess/iio-sensor-proxy #216: Support tent and tablet mode orientation sensors

dos1, 2018-02-23:
Some laptops, especially laptop-convertibles with 360° hinges like Lenovo Yoga series, usually contain a pair of two sensors: an accelerometer mounted at the screen, and a gyroscope or a second accelerometer mounted at the keyboard.

iio-sensor-proxy could be extended to also provide the second orientation value, so use-cases like tent or tablet mode 

It also has to be made sure that iio-sensor-proxy uses correct accelerometer for screen orientation purpose (not sure if it does already - my laptop has a gyro instead of 2nd accel). A case where there's only gyro(s) and no accels could also be considered - not sure if there's some hardware like that though.

hadess, 2018-02-24:
Detecting tablet mode, or tent mode, is out of scope (and would be utterly unreliable if we did it based on the values of the accelerometers), and should be handled in the kernel and libinput, so that the keyboard and touchpad can be disabled appropriately.

About reporting the accelerometer reading values, see #166

closed 2018-02-24 by hadess

dos1, 2018-02-24:
I don't think that's right. While the 100%-keyboard-behind-the-screen tablet mode is indeed kernel/libinput thing, as it's usually based on hall sensor similarly to the lid switch (and there is some code for handling it already there), in some devices there's also a full range of motion available with so-called tent and presentation modes that has to be detected using values from multiple accelerometers (/gyros/inclinometers/etc.) and is definitely out of kernel and libinput scope.

I probably made a mistake by mentioning the tablet mode as an example, just thought it's the simplest one :P

But yeah, #166 seems like the interesting one.

dos1, 2018-02-24:
Yes, the BIOS even turns off the keyboard and its backlight automatically for me without Linux noticing at all. However, that doesn't use accelerometers at all, at least on my device, and is only a binary switch that detects whether the tablet mode is fully engaged or not (and it can be toggled with a magnet put into right place).

For other modes, you have to use two accelerometers, since otherwise when the screen stays in the same position and only the keyboard moves (but doesn't close completely on either of the screen sides), it's undetectable, and yet you might want to disable the keyboard and touchpad when the laptop is standing on them.

hadess, 2018-03-05:
See: https://lkml.org/lkml/2017/9/15/188