hadess/iio-sensor-proxy #218: On Dell 5855 (and possibly also on 5056) iio-sensor-proxy read accelerometer measurements only at launch

RussianNeuroMancer, 2018-03-08:
On Dell 5855 (and possibly also on 5056) iio-sensor-proxy read accelerometer measurements only once at launch, so tablet rotation after OS launch is not detected (as I can see by monitor-sensor output). However, if something (like [this script](http://www.studioteabag.com/science/dell-venue-pro-linux/#fun_with_iio)) is constantly polling sysfs in background, iio-sensor-proxy is able to read new values and report orientation to DE accordingly. 

Issue is reproducible for at least 4.8-4.16 range, maybe earlier. Re-tested it today with Linux 4.16rc4.

hadess, 2018-03-09:
Yep, the kernel broke again, see #200

RussianNeuroMancer, 2019-01-14:
This issue is no longer reproducible on Ubuntu 18.10 with Linux 4.18.0.

closed 2019-01-14 by RussianNeuroMancer