hadess/iio-sensor-proxy #219: HP Spectre x360: Screen and touchscreen rotate - pen doesn't rotate.

jcfergus, 2018-03-15:
I'm not sure if this is an iio-sensor-proxy issue or a GNOME issue (or a user issue).  I'm running Ubuntu 17.10.1 on a HP Spectre x360, GNOME on Wayland, kernel 4.13.0-36.  iio-sensor-proxy is correctly detecting the sensor and rotating the screen and touchscreen inputs, but regardless of the screen rotation the pen input coordinates stay in the original orientation.   Any suggestions?  Details below.  

The pen shows as a second mouse, and is not recognized by xsetwacom .  

```
jferg@quartz ~ $ xinput -list --long
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]

⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer:13                     	id=6	[slave  pointer  (2)]
⎜   ↳ xwayland-relative-pointer:13            	id=7	[slave  pointer  (
⎜   ↳ xwayland-touch:13                       	id=9	[slave  pointer  (2)]
⎜   ↳ xwayland-stylus:13                      	id=10	[slave  pointer  (2)]
	Reporting 7 classes:
		Class originated from: 10. Type: XIButtonClass
		Buttons supported: 9
		Button labels: None None None None None None None None None
		Button state:
		Class originated from: 10. Type: XIValuatorClass
		Detail for Valuator 0:
		  Label: Abs X
		  Range: 0.000000 - 262143.000000
		  Resolution: 10000 units/m
		  Mode: absolute
		  Current value: 80964.000000
		Class originated from: 10. Type: XIValuatorClass
		Detail for Valuator 1:
		  Label: Abs Y
		  Range: 0.000000 - 262143.000000
		  Resolution: 10000 units/m
		  Mode: absolute
		  Current value: 151582.000000
		Class originated from: 10. Type: XIValuatorClass
		Detail for Valuator 2:
		  Label: Abs Pressure
		  Range: 0.000000 - 65535.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 10. Type: XIValuatorClass
		Detail for Valuator 3:
		  Label: Abs Tilt X
		  Range: -64.000000 - 63.000000
		  Resolution: 57 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 10. Type: XIValuatorClass
		Detail for Valuator 4:
		  Label: Abs Tilt Y
		  Range: -64.000000 - 63.000000
		  Resolution: 57 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 10. Type: XIValuatorClass
		Detail for Valuator 5:
		  Label: Abs Wheel
		  Range: -900.000000 - 899.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000

⎜   ↳ xwayland-eraser:13                      	id=11	[slave  pointer  (2)]
⎜   ↳ xwayland-cursor:13                      	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ xwayland-keyboard:13                    	id=8	[slave  keyboard (3)]
```

```
$ udevadm info /dev/input/mouse2
P: /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-6/i2c-ELAN0732:00/0018:04F3:22E2.0005/input/input21/mouse2
N: input/mouse2
S: input/by-path/pci-0000:00:15.0-platform-i2c_designware.0-mouse
E: DEVLINKS=/dev/input/by-path/pci-0000:00:15.0-platform-i2c_designware.0-mouse
E: DEVNAME=/dev/input/mouse2
E: DEVPATH=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-6/i2c-ELAN0732:00/0018:04F3:22E2.0005/input/input21/mouse2
E: ID_INPUT=1
E: ID_INPUT_TABLET=1
E: ID_PATH=pci-0000:00:15.0-platform-i2c_designware.0
E: ID_PATH_TAG=pci-0000_00_15_0-platform-i2c_designware_0
E: ID_SERIAL=noserial
E: MAJOR=13
E: MINOR=34
E: SUBSYSTEM=input
E: USEC_INITIALIZED=9683499
```

jcfergus, 2018-03-16:
Ok, so the following config got me working:  

```
hpspectrex360.tablet:
[Device]
Name=ELAN0732:00 04F3:22E2 Pen
DeviceMatch=i2c:04f3:22e2
IntegratedIn=Display;System
Styli=0xfffff;0xffffe

[Features]
Stylus=true
Reversible=false
Touch=false
TouchSwitch=false
Ring=false
Ring2=false
NumStrips=0
Buttons=0
```

The stylus still doesn't show up in gnome-control-center, but the tablet does, and rotation works now.

hadess, 2018-05-14:
As mentioned in the README, rotation is handled in the desktop environment. We just read sensors.

closed 2018-05-14 by hadess

sm8ps, 2019-01-10:
Hi Justin
I have the same problem and would like to try your solution but I do not see where to put the configuration file. It certainly does not look like an Xorg.conf to me. Would you please explain? Thanks in advance!
PS: Thanks to everybody responsible for providing iio-sensor-proxy! It is a really, really useful tool.