hadess/iio-sensor-proxy #220: Constant orientation updates

aik, 2018-03-16:
Tried this on Lenovo Yoga 710-11ISK - it keeps rotating the screen as the monitor-sensor keeps reporting changes. Fedora 27, iio-sensor-proxy has been compiled 15min ago, is that a broken sensor?

```
[root@aikyoga iio-sensor-proxy]# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:3420): DEBUG: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3
** (process:3420): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:3420): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/trigger4
** (process:3420): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4
** (process:3420): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4 of type accelerometer at IIO Buffer accelerometer
** (process:3420): DEBUG: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device2
** (process:3420): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:3420): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/trigger4
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/scan_elements/in_accel_y_en
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/scan_elements/in_accel_x_en
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/scan_elements/in_timestamp_en
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/scan_elements/in_accel_z_en
** (process:3420): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3420): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3420): DEBUG: Got type for in_timestamp: is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:3420): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3420): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3420): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3420): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3420): DEBUG: Built channel array for in_timestamp: index: 3, is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:3420): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200041.9.auto/trigger3
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3/scan_elements/in_illuminance_en
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3/scan_elements/in_intensity_both_en
** (process:3420): DEBUG: Got type for in_illuminance: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Built channel array for in_illuminance: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/trigger2
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device2/scan_elements/in_magn_z_en
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device2/scan_elements/in_magn_y_en
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device2/scan_elements/in_rot_from_north_magnetic_tilt_comp_en
** (process:3420): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-4/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device2/scan_elements/in_magn_x_en
** (process:3420): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3420): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3420): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3420): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3420): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3420): DEBUG: No new data available on 'iio:device3'
** (process:3420): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
** (process:3420): DEBUG: process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
** (process:3420): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
** (process:3420): DEBUG: Accel read from IIO on 'iio:device4': -31, -859, -503 (scale 0.009807)
** (process:3420): DEBUG: Accel sent by driver (quirk applied): -31, -859, -503 (scale: 0.009807)
** (process:3420): DEBUG: Emitted orientation changed: from undefined to normal
** (process:3420): DEBUG: No new data available on 'iio:device3'
** (process:3420): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
** (process:3420): DEBUG: process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
** (process:3420): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
** (process:3420): DEBUG: Accel read from IIO on 'iio:device4': 1607, 0, 23988 (scale 0.009807)
[... cut ...]
** (process:3420): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
** (process:3420): DEBUG: process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
** (process:3420): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
** (process:3420): DEBUG: Accel read from IIO on 'iio:device4': 0, 0, 0 (scale 0.009807)
** (process:3420): DEBUG: Accel sent by driver (quirk applied): 0, 0, 0 (scale: 0.009807)
^C
[root@aikyoga iio-sensor-proxy]# 

```

hadess, 2018-05-14:
Stop iio-sensor-proxy, and try reading the values directly from the sysfs files. The values certainly seem weird.

aik, 2018-06-17:
Recorded some raw readings in process_scan_1():
https://github.com/aik/iio-sensor-proxy/commit/be6ee47932e2455ef2cafef02ea754a20b8521a7

** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': -15952, -28810, 20774 (scale 0.009807)  8f76c1b0 51268f76 5126
** (process:22741): WARNING **: Accel sent by driver (quirk applied): -15952, -28810, 20774 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from bottom-up to normal
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 0, 0, 0 (scale 0.009807)  0 0 0
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 0, 0, 0 (scale: 0.009807)
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 0, 0, 0 (scale 0.009807)  0 0 0
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 0, 0, 0 (scale: 0.009807)
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 0, 0, -22381 (scale: 0.009807)
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 0, 0, 0 (scale 0.009807)  0 0 0
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 0, 0, 0 (scale: 0.009807)
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 0, -22381, 20774 (scale 0.009807)  a8930000 5126a893 5126
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 0, -22381, 20774 (scale: 0.009807)
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': -32146, -6309, -14224 (scale 0.009807)  e75b826e c870e75b ff00c870
** (process:22741): WARNING **: Accel sent by driver (quirk applied): -32146, -6309, -14224 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from normal to right-up
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 9039, -6309, -6550 (scale 0.009807)  e75b234f e66ae75b ff00e66a
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 9039, -6309, -6550 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from right-up to left-up
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': -17680, -13982, 32146 (scale 0.009807)  c962baf0 7d92c962 ff007d92
** (process:22741): WARNING **: Accel sent by driver (quirk applied): -17680, -13982, 32146 (scale: 0.009807)
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': -15831, 27082, -15952 (scale 0.009807)  69cac229 c1b069ca c1b0
** (process:22741): WARNING **: Accel sent by driver (quirk applied): -15831, 27082, -15952 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from left-up to bottom-up
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 12858, 13100, 242 (scale 0.009807)  332c323a f2332c 650000f2
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 12858, 13100, 242 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from bottom-up to left-up
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': -28447, -27203, -3214 (scale 0.009807)  95bd90e1 f37295bd ff00f372
** (process:22741): WARNING **: Accel sent by driver (quirk applied): -28447, -27203, -3214 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from left-up to right-up
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 15, -878, -480 (scale 0.009807)  fc92000f fe20fc92 fe20
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 15, -878, -480 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from right-up to normal
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 0, 0, 0 (scale 0.009807)  0 0 0
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 0, 0, 0 (scale: 0.009807)
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 0, 0, 0 (scale 0.009807)  0 0 0
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 0, 0, 0 (scale: 0.009807)
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 28690, 0, 6429 (scale 0.009807)  7012 191d0000 ff00191d
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 28690, 0, 6429 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from normal to left-up
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 1487, 30417, -19167 (scale 0.009807)  76d105cf b52176d1 ff00b521
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 1487, 30417, -19167 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from left-up to bottom-up
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': -14465, -19167, -22381 (scale 0.009807)  b521c77f a893b521 ff00a893
** (process:22741): WARNING **: Accel sent by driver (quirk applied): -14465, -19167, -22381 (scale: 0.009807)
** (process:22741): WARNING **: Emitted orientation changed: from bottom-up to normal
** (process:22741): WARNING **: BBBB Accel read from IIO on 'iio:device0': 0, 0, 0 (scale 0.009807)  0 0 ff000000
** (process:22741): WARNING **: Accel sent by driver (quirk applied): 0, 0, 0 (scale: 0.009807)

This numbers came when I just moved a laptop on a desk, no turning, just moving left/right by 1cm. The numbers look dodgy though.

hadess, 2018-06-17:
Either the driver or the device seems to send garbage data. This is assuming there isn't a bug in iio-sensor-proxy for reading the update events from the buffered accelerometer. You should send a mail to the linux-iio mailing-list about it.

As for the 90 degrees rotation, see https://github.com/hadess/iio-sensor-proxy/#accelerometer-orientation

aik, 2018-06-17:
I have a suspicion that the sensor might be broke actually as the screen does not even try to change screen  orientation when I hold laptop in tablet mode (i.e. folded) in front of me and then turn it 90deg clockwise but it does when I turn 90deg clockcounterwise, is there a way to tell if it is not broken?

hadess, 2018-06-18:
Look at the driver first.

aik, 2019-03-16:
So. I got bored and looked there again :)This time it is fedora29 v4.20.14. The problem is there with the stock package. Then I tried git version of iio-sensor-proxy, same issue. Then I comment out &iio_buffer_accel to make &iio_poll_accel work - and things worked nicely. I looked in sysfs and  in_accel_?_raw seem to have correct values (same as in the first log below, give or take), all good. Recorded some debug from iio-sensor-proxy:
```
Accel read from IIO on 'accel_3d': -39, -937, -378 (scale 0.009807)
Accel sent by driver (quirk applied): -39, -937, -378 (scale: 0.009807)
Emitted orientation changed: from undefined to normal
No new data available on 'iio:device3'
Accel read from IIO on 'accel_3d': -39, -933, -371 (scale 0.009807)
Accel sent by driver (quirk applied): -39, -933, -371 (scale: 0.009807)
No new data available on 'iio:device3'
Accel read from IIO on 'accel_3d': -39, -933, -367 (scale 0.009807)
Accel sent by driver (quirk applied): -39, -933, -367 (scale: 0.009807)
```
This is the good log, gnome works fine.

Then I recorded debug with the buffer driver enabled:
```
rocess_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
Accel read from IIO on 'iio:device4': -15, -898, -375 (scale 0.009807)
Accel sent by driver (quirk applied): -15, -898, -375 (scale: 0.009807)
Emitted orientation changed: from undefined to normal
No new data available on 'iio:device3'
process_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
Accel read from IIO on 'iio:device4': 20774, 27203, 0 (scale 0.009807)
Accel sent by driver (quirk applied): 20774, 27203, 0 (scale: 0.009807)
Emitted orientation changed: from normal to left-up
No new data available on 'iio:device3'
process_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
Accel read from IIO on 'iio:device4': -31, -929, -398 (scale 0.009807)
Accel sent by driver (quirk applied): -31, -929, -398 (scale: 0.009807)
Emitted orientation changed: from left-up to normal
No new data available on 'iio:device3'
process_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
Accel read from IIO on 'iio:device4': -14345, -32024, 12738 (scale 0.009807)
Accel sent by driver (quirk applied): -14345, -32024, 12738 (scale: 0.009807)
```
So it is good reading, bad reading, good reading, bad reading, and gnome rotates the screen non stop. I tend to blame the driver then but the IIO framework in the driver is quite complex and I failed to find quickly where there reads are handled in the kernel. Any pointers?

Here is `dmesg | grep i2c`:
```
[root@aikyoga iio:device4]# dmesg | egrep '(i2c|iio)'
[    5.389867] i2c_hid i2c-ITE8350:00: i2c-ITE8350:00 supply vdd not found, using dummy regulator
[    5.389893] i2c_hid i2c-ITE8350:00: Linked as a consumer to regulator.0
[    5.389896] i2c_hid i2c-ITE8350:00: i2c-ITE8350:00 supply vddl not found, using dummy regulator
[    5.502896] hid-generic 0018:048D:8350.0002: hidraw1: I2C HID v1.00 Device [ITE8350:00 048D:8350] on i2c-ITE8350:00
[    5.528455] i2c_hid i2c-SYNA2B23:00: i2c-SYNA2B23:00 supply vdd not found, using dummy regulator
[    5.528485] i2c_hid i2c-SYNA2B23:00: Linked as a consumer to regulator.0
[    5.528489] i2c_hid i2c-SYNA2B23:00: i2c-SYNA2B23:00 supply vddl not found, using dummy regulator
[    5.543440] input: SYNA2B23:00 06CB:2714 Mouse as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-6/i2c-SYNA2B23:00/0018:06CB:2714.0003/input/input13
[    5.543690] hid-generic 0018:06CB:2714.0003: input,hidraw2: I2C HID v1.00 Mouse [SYNA2B23:00 06CB:2714] on i2c-SYNA2B23:00
[    6.053237] input: Synaptics TM2714-002 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-6/i2c-SYNA2B23:00/0018:06CB:2714.0003/input/input16
[    6.053444] hid-rmi 0018:06CB:2714.0003: input,hidraw1: I2C HID v1.00 Mouse [SYNA2B23:00 06CB:2714] on i2c-SYNA2B23:00
```

hadess, 2019-03-16:
I've added more debug in git master, if you want to try again. The additional info might show whether other bits of data change.

As for the IIO subsystem, the code lives in `drivers/iio/accel/`.

aik, 2019-03-16:
I found drivers/iio/accel but could not find the sysfs node for in_accel_x_raw, for example, it is hidded behind macros or comes from outside the kernel, what is it?

Anyway, here is a log with the new "bits used":

```[root@aikyoga src]# G_MESSAGES_DEBUG=all ./iio-sensor-proxy
Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3
Found device /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
Found associated trigger at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/trigger4
Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4
Found device /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4 of type accelerometer at IIO Buffer accelerometer
Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device5
Found device /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device5 of type compass at IIO Buffer Compass
Found associated trigger at /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/trigger4
Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/scan_elements/in_accel_x_en
Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/scan_elements/in_accel_z_en
Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/scan_elements/in_timestamp_en
Already enabled sensor /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/scan_elements/in_accel_y_en
Trying to read 'in_accel_x_scale' (name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/in_accel_x_scale: Success
Trying to read 'in_accel_scale' (generic name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Trying to read 'in_accel_x_offset' (name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/in_accel_x_offset: Success
Trying to read 'in_accel_offset' (generic name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Trying to read 'in_accel_z_scale' (name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/in_accel_z_scale: Success
Trying to read 'in_accel_scale' (generic name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Trying to read 'in_accel_z_offset' (name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/in_accel_z_offset: Success
Trying to read 'in_accel_offset' (generic name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Trying to read 'in_timestamp_scale' (name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4/in_timestamp_scale: Success
Trying to read 'in_scale' (generic name) from dir '/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device4'
[... cut ...]
Emitted orientation changed: from left-up to normal
No new data available on 'iio:device3'
process_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
A Accel read from IIO on 'iio:device4': -14345, -32024, 12738 (scale 0.009807)
Accel sent by driver (quirk applied): -14345, -32024, 12738 (scale: 0.009807)
```

hadess, 2019-03-21:
The problem isn't limited to the accelerometer:
```
Emitted light changed: from 0.000000 to 2249.999950
Emitted light changed: from 2249.999950 to 6.110000
Emitted light changed: from 6.110000 to 6.040000
```

The first 2 changes are pretty bad.

Does this change make any difference on your system?
```patch
diff --git a/src/iio-buffer-utils.c b/src/iio-buffer-utils.c
index 4818d3e..9eb3fcd 100644
--- a/src/iio-buffer-utils.c
+++ b/src/iio-buffer-utils.c
@@ -68,7 +68,7 @@ iio_readu32 (struct iio_channel_info *info, const guint8 * data)
 }
 
 #define _IIO_GET(__data, __idx, __size, __shift) \
-    (((guint##__size) (((const guint8 *) (__data))[__idx])) << (__shift))
+    (((gint##__size) (((const gint8 *) (__data))[__idx])) << (__shift))
 
 #define _IIO_READ_INT32_BE(data)       (_IIO_GET (data, 0, 32, 24) | \
                                         _IIO_GET (data, 1, 32, 16) | \
```

I'm wondering why you saw `0.0 0.0 0.0` readings using `iio_generic_buffer` and not using `iio-sensor-proxy`.

aik, 2019-03-21:
No luck there. 
[iiologbuffer.txt](https://github.com/hadess/iio-sensor-proxy/files/2991669/iiologbuffer.txt)
[iiologpolling.txt](https://github.com/hadess/iio-sensor-proxy/files/2991670/iiologpolling.txt)

hadess, 2019-05-13:
I've made some pretty big changes to the code to bring it back in line with the upstream code it was derived from 6 or 7 years ago (iio-sensor-proxy was based on a fork of a fork).

See the PR here for testing: https://github.com/hadess/iio-sensor-proxy/pull/269

The end goal here is that the `iio_generic_buffer` tool and iio-sensor-proxy give out the same readings. Anything else we can deal with afterwards.

hadess, 2019-05-14:
Please test with version 2.6.

aik, 2019-05-24:
Sorry for the delay but it is the exact same behaviour - keeps rotating, just seems slightly quicker than before. I am attaching the log. The laptop was on my laps and just a little movement make screen spinning like crazy.

[iioproxylog_2_7.txt](https://github.com/hadess/iio-sensor-proxy/files/3216876/iioproxylog_2_7.txt)

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/220

closed 2020-01-09 by hadess