hadess/iio-sensor-proxy #221: [Feature] Support extra sensors

Depau, 2018-03-19:
I just noticed that my computer (Lenovo ThinkPad X1 Yoga 1st) has also a rotation and magnetic field sensor:

```shell
$ cat /sys/bus/iio/devices/iio:*/name
gyro_3d
magn_3d
gyro_3d
accel_3d
als
```

It would be nice if IIO sensor proxy supported them too, so that new apps can be developed on top of it (I would easily make a compass app just for fun).

hadess, 2018-05-14:
Compasses are already supported through Geoclue. There's no direct access to the compass.

For what it's worth:
> gyro_3d

A gyro which we don't need (or would use in combination with the accelerometer)

> magn_3d

Already used by Geoclue

> gyro_3d

Another gyro.

> accel_3d

The accelerometer.

> als

Light sensors, which are also already supported.

closed 2018-05-14 by hadess

Depau, 2018-05-14:
Thank you.
May I ask why you're not going to support gyro? I like the current dbus API for the other sensors.
Is this project strictly aiming at what may be useful for a desktop environment?

hadess, 2018-05-14:
> May I ask why you're not going to support gyro?

Why do you want to use it? If it's something we're interested in, we'll probably accept patches for it.

> I like the current dbus API for the other sensors.
> Is this project strictly aiming at what may be useful for a desktop environment?

Depends on what you want to use it for.