hadess/iio-sensor-proxy #224: After upgrade to Linux 4.17rc1 it's required to suspend/resume to get accelerometer and ambient light sensor working

RussianNeuroMancer, 2018-04-16:
Accelerometer and ambient light sensor works only after suspend/resume on Lenovo Miix2 8 after upgrade from Linux 4.16 to Linux 4.17rc1. Both sensors is detected after boot but undefined:

```
~$ monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
```
After suspend/resume both sensors start working normally.

There is no such issue with Linux 4.16.0.

hadess, 2018-05-14:
It's a kernel issue, nothing I can do about it.

MCSH, 2018-05-19:
I'm not sure if it's the same problem but after suspend/resume when it rotates, the screen goes blank and only the cursor is visible

RussianNeuroMancer, 2018-05-19:
@hadess 

> It's a kernel issue, nothing I can do about it.

Ok, I'll recheck with 4.17 and fill to kernel bugtracker.

@MCSH 

> the screen goes blank and only the cursor is visible

Your issue is either combination of two (this one and some other in video driver, compositor, something else) or completely different.

closed 2018-05-19 by RussianNeuroMancer