hadess/iio-sensor-proxy #227: monitor-sensor shows only data about light

sorawee, 2018-05-13:
Hi,

I just switched to Linux Mint 18.3 with the kernel `4.13.0-41-generic`. Running `monitor-sensor`, I saw:

```
$ monitor-sensor
** Message: Light changed: 3.892000 (lux)
** Message: Light changed: 4.022000 (lux)
** Message: Light changed: 3.373000 (lux)
** Message: Light changed: 2.465000 (lux)
** Message: Light changed: 2.335000 (lux)
```

Note that the lines:

```
Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer ...
```

didn't even appear!

Just in case this information is useful:

```
$ G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:4195): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device4
** (process:4195): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device4 of type ambient light sensor at IIO Buffer Light sensor
** (process:4195): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3
** (process:4195): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:4195): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device1
** (process:4195): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device1 of type compass at IIO Buffer Compass
** (process:4195): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/trigger3

** (process:4195): WARNING **: Failed to write current_trigger file Permission denied

** (process:4195): WARNING **: Could not open for write '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3/buffer/enable'
** (process:4195): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/trigger4

** (process:4195): WARNING **: Failed to write current_trigger file Permission denied

** (process:4195): WARNING **: Could not open for write '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device4/buffer/enable'
** (process:4195): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/trigger1

** (process:4195): WARNING **: Failed to write current_trigger file Permission denied

** (process:4195): WARNING **: Could not open for write '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device1/buffer/enable'
```

So... how can I make `monitor-sensor` shows data from accelerometer properly? Thanks!

hadess, 2018-05-13:
I don't know, but if you don't run it as root, it's certainly not going to work.

sorawee, 2018-05-13:
I did run it as root, and the result for `monitor-sensor` is the same.

hadess, 2018-05-13:
I'd need the output of iio-sensor-proxy itself, but run as root, there's tons of permission errors in your logs.

sorawee, 2018-05-14:
Yeah, my original issue didn't run `iio-sensor-proxy` as root, but I realized that and fixed it 6 hours ago. It's weird that you still see the old one. Anyway, here's the the output when run as root.

```
# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:4472): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device4
** (process:4472): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device4 of type ambient light sensor at IIO Buffer Light sensor
** (process:4472): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3
** (process:4472): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:4472): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device1
** (process:4472): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device1 of type compass at IIO Buffer Compass
** (process:4472): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/trigger3
** (process:4472): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

** (process:4472): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3/in_scale

** (process:4472): WARNING **: Problem reading scan element information: /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3
** (process:4472): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/trigger4
** (process:4472): DEBUG: Got type for in_illuminance: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Built channel array for in_illuminance: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/trigger1
** (process:4472): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4472): DEBUG: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.
```

hadess, 2018-05-14:
What does a `cat` of `/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3/in_scale` give you?

> Yeah, my original issue didn't run iio-sensor-proxy as root, but I realized that and fixed it 6 hours ago.

I read from the mail notifications. How am I to know that you fixed that problem in your comment if you don't say so?

sorawee, 2018-05-14:
> I read from the mail notifications. How am I to know that you fixed that problem in your comment if you don't say so?

I see. Sorry about that.

> What does a cat of `/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3/in_scale` give you?

It doesn't exist.

```
# cd /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3/
# ls
buffer                   dev                  in_accel_offset              in_accel_scale  in_accel_y_raw  name   scan_elements  trigger
current_timestamp_clock  in_accel_hysteresis  in_accel_sampling_frequency  in_accel_x_raw  in_accel_z_raw  power  subsystem      uevent
```

hadess, 2018-05-14:
You seem to be using iio-sensor-proxy 2.2 or older. Could you please try with a newer version?

sorawee, 2018-05-15:
Yup, resolved :) Thanks!

closed 2018-05-15 by sorawee