hadess/iio-sensor-proxy #228: No Auto Rotation of Screen Orientation

natedawg92, 2018-05-31:
## Setup
Setting | Value
--- | ---
Device | Asus Transformer Book T100HA
OS | Ubuntu 18.04
Kernel | 4.15.0-22-generic
accelerometer | HID-SENSOR-200073.6.auto
Compass | HID-SENSOR-200083.28.auto
Ambient Light | N/A

## DEBUG
```BASH
$ G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:2709): DEBUG: 00:48:11.496: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.6.auto/trigger0
** (process:2709): DEBUG: 00:48:11.497: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.6.auto/iio:device0
** (process:2709): DEBUG: 00:48:11.497: Found device /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.6.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:2709): DEBUG: 00:48:11.501: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0007/HID-SENSOR-200083.28.auto/iio:device8
** (process:2709): DEBUG: 00:48:11.501: Found device /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0007/HID-SENSOR-200083.28.auto/iio:device8 of type compass at IIO Buffer Compass
** (process:2709): DEBUG: 00:48:11.604: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.6.auto/trigger0
** (process:2709): DEBUG: 00:48:11.605: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.6.auto/iio:device0/scan_elements/in_accel_y_en
** (process:2709): DEBUG: 00:48:11.606: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.6.auto/iio:device0/scan_elements/in_accel_x_en
** (process:2709): DEBUG: 00:48:11.606: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.6.auto/iio:device0/scan_elements/in_timestamp_en
** (process:2709): DEBUG: 00:48:11.607: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.6.auto/iio:device0/scan_elements/in_accel_z_en
** (process:2709): DEBUG: 00:48:11.609: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.610: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.611: Got type for in_timestamp: is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:2709): DEBUG: 00:48:11.612: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.613: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.613: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.613: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.613: Built channel array for in_timestamp: index: 3, is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:2709): DEBUG: 00:48:11.652: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0007/HID-SENSOR-200083.28.auto/trigger8
** (process:2709): DEBUG: 00:48:11.653: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0007/HID-SENSOR-200083.28.auto/iio:device8/scan_elements/in_magn_z_en
** (process:2709): DEBUG: 00:48:11.654: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0007/HID-SENSOR-200083.28.auto/iio:device8/scan_elements/in_magn_y_en
** (process:2709): DEBUG: 00:48:11.654: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0007/HID-SENSOR-200083.28.auto/iio:device8/scan_elements/in_rot_from_north_magnetic_tilt_comp_en
** (process:2709): DEBUG: 00:48:11.655: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0007/HID-SENSOR-200083.28.auto/iio:device8/scan_elements/in_magn_x_en
** (process:2709): DEBUG: 00:48:11.657: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.658: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.659: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.660: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.661: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.661: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.661: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.661: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2709): DEBUG: 00:48:11.677: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2709): DEBUG: 00:48:11.678: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2709): DEBUG: 00:48:11.706: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2709): DEBUG: 00:48:12.380: No new data available on 'iio:device0'
** (process:2709): DEBUG: 00:48:13.081: No new data available on 'iio:device0'
** (process:2709): DEBUG: 00:48:13.783: No new data available on 'iio:device0'
```
As suggested on previous issues, running `watch -n 0.1 cat '/sys/bus/iio/devices/iio:device'*/*raw*` enables the sensor to work as expected and changes the orientation as the screen is rotated. 

If this is a duplicate of an existing issue please close, just wasn't too sure if this was an exact duplicate. if you require any further information let me know and ill get it to you ASAP.

hadess, 2018-06-01:
The first thing to try would be a newer kernel. The sensor drivers keep breaking.

natedawg92, 2018-06-01:
I did try upgrading the kernel to 4.16.13 but then couldn't boot, honestly not too clued up on OS level software.

natedawg92, 2018-06-03:
@hadess Tried  all Kernels between 4.15.0 and 4.16.13 none worked for auto rotation. I could try downgrading but as i said previously I don't know too much so will look at what that may affect before doing anything.