hadess/iio-sensor-proxy #231: iio-sensor-proxy makes my load average == 1.0

ljanvier, 2018-06-30:
Hi,

I have a lot of of ProBook 650/450 with Ubuntu 18.04. They have a "ST LIS3LV02DL Accelerometer".

If iio-sensor-proxy is running and has a client (gdm or monitor-sensor running), the load average is =~ 1.0, even if the idle > 99%. iio-sensor-proxy is always in D state (uninterruptible sleep). In DEBUG mode, iio-sensor-proxy shows a message every 2 seconds (see below).

With iio-sensor-proxy stopped or without clients, the load average == 0.0.

Not sure if it's a iio-sensor-proxy problem or a driver problem.

Debug info:

~~~
Linux 4.18.0-041800rc2-generic #201806241430 SMP Sun Jun 24 14:33:07 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

iio-sensor-proxy Version: 2.4-2

# monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor

# gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};

# udevadm info --export-db|grep -10 -i iio

P: /devices/platform/lis3lv02d/input/input26
E: ABS=7
E: DEVPATH=/devices/platform/lis3lv02d/input/input26
E: EV=9
E: ID_FOR_SEAT=input-platform-lis3lv02d
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_PATH=platform-lis3lv02d
E: ID_PATH_TAG=platform-lis3lv02d
E: IIO_SENSOR_PROXY_TYPE=input-accel
E: MODALIAS=input:b0019v0000p0000e0000-e0,3,kra0,1,2,mlsfw
E: NAME="ST LIS3LV02DL Accelerometer"
E: PHYS="lis3lv02d/input0"
E: PRODUCT=19/0/0/0
E: PROP=0
E: SUBSYSTEM=input
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:seat:
E: USEC_INITIALIZED=10810791

P: /devices/platform/lis3lv02d/input/input26/event19
N: input/event19
S: input/by-path/platform-lis3lv02d-event
E: DEVLINKS=/dev/input/by-path/platform-lis3lv02d-event
E: DEVNAME=/dev/input/event19
E: DEVPATH=/devices/platform/lis3lv02d/input/input26/event19
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_PATH=platform-lis3lv02d
E: ID_PATH_TAG=platform-lis3lv02d
E: IIO_SENSOR_PROXY_TYPE=input-accel
E: LIBINPUT_DEVICE_GROUP=19/0/0:lis3lv02d
E: MAJOR=13
E: MINOR=83
E: SUBSYSTEM=input
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=11687230

P: /devices/platform/lis3lv02d/input/input26/js0
N: input/js0
E: DEVNAME=/dev/input/js0
E: DEVPATH=/devices/platform/lis3lv02d/input/input26/js0
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_PATH=platform-lis3lv02d
E: ID_PATH_TAG=platform-lis3lv02d
E: IIO_SENSOR_PROXY_TYPE=input-accel
E: MAJOR=13
E: MINOR=0
E: SUBSYSTEM=input
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=10834283

P: /devices/platform/microcode
E: DEVPATH=/devices/platform/microcode
E: MODALIAS=platform:microcode
E: SUBSYSTEM=platform

P: /devices/platform/pcspkr
E: DEVPATH=/devices/platform/pcspkr

G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:28958): DEBUG: 08:43:27.813: Found input accel at /sys/devices/platform/lis3lv02d/input/input26/event19
** (process:28958): DEBUG: 08:43:27.814: Found device /sys/devices/platform/lis3lv02d/input/input26/event19 of type accelerometer at Input accelerometer
** (process:28958): DEBUG: 08:43:28.655: Accel read from input on '(null)': -24, -18, 1158 (scale 0.038320)
** (process:28958): DEBUG: 08:43:28.655: Accel sent by driver (quirk applied): -24, -18, 1158 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:32.938: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:28958): DEBUG: 08:44:34.487: Accel read from input on '(null)': -13, 9, 1145 (scale 0.038320)
** (process:28958): DEBUG: 08:44:34.488: Accel sent by driver (quirk applied): -13, 9, 1145 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:35.323: Accel read from input on '(null)': -24, 7, 1157 (scale 0.038320)
** (process:28958): DEBUG: 08:44:35.323: Accel sent by driver (quirk applied): -24, 7, 1157 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:36.163: Accel read from input on '(null)': -29, 6, 1176 (scale 0.038320)
** (process:28958): DEBUG: 08:44:36.163: Accel sent by driver (quirk applied): -29, 6, 1176 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:36.999: Accel read from input on '(null)': -26, 22, 1176 (scale 0.038320)
** (process:28958): DEBUG: 08:44:36.999: Accel sent by driver (quirk applied): -26, 22, 1176 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:37.839: Accel read from input on '(null)': -26, 22, 1164 (scale 0.038320)
** (process:28958): DEBUG: 08:44:37.839: Accel sent by driver (quirk applied): -26, 22, 1164 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:37.839: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:28958): DEBUG: 08:44:38.683: Accel read from input on '(null)': -25, 11, 1164 (scale 0.038320)
** (process:28958): DEBUG: 08:44:38.683: Accel sent by driver (quirk applied): -25, 11, 1164 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:39.523: Accel read from input on '(null)': -24, 22, 1116 (scale 0.038320)
** (process:28958): DEBUG: 08:44:39.523: Accel sent by driver (quirk applied): -24, 22, 1116 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:40.359: Accel read from input on '(null)': 0, 8, 1170 (scale 0.038320)
** (process:28958): DEBUG: 08:44:40.359: Accel sent by driver (quirk applied): 0, 8, 1170 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:41.199: Accel read from input on '(null)': -22, 22, 1163 (scale 0.038320)
** (process:28958): DEBUG: 08:44:41.199: Accel sent by driver (quirk applied): -22, 22, 1163 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:42.043: Accel read from input on '(null)': -25, 12, 1148 (scale 0.038320)
** (process:28958): DEBUG: 08:44:42.043: Accel sent by driver (quirk applied): -25, 12, 1148 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:42.883: Accel read from input on '(null)': -24, 11, 1159 (scale 0.038320)
** (process:28958): DEBUG: 08:44:42.883: Accel sent by driver (quirk applied): -24, 11, 1159 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:43.723: Accel read from input on '(null)': -24, 8, 1161 (scale 0.038320)
** (process:28958): DEBUG: 08:44:43.723: Accel sent by driver (quirk applied): -24, 8, 1161 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:44.563: Accel read from input on '(null)': -27, 8, 1158 (scale 0.038320)
** (process:28958): DEBUG: 08:44:44.563: Accel sent by driver (quirk applied): -27, 8, 1158 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:45.399: Accel read from input on '(null)': -29, 23, 1163 (scale 0.038320)
** (process:28958): DEBUG: 08:44:45.399: Accel sent by driver (quirk applied): -29, 23, 1163 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:46.235: Accel read from input on '(null)': -26, 23, 1175 (scale 0.038320)
** (process:28958): DEBUG: 08:44:46.235: Accel sent by driver (quirk applied): -26, 23, 1175 (scale: 0.038320)
** (process:28958): DEBUG: 08:44:47.067: Accel read from input on '(null)': -13, 23, 1160 (scale 0.038320)
** (process:28958): DEBUG: 08:44:47.067: Accel sent by driver (quirk applied): -13, 23, 1160 (scale: 0.038320)

czirkoszoltan, 2018-07-17:
I confirm this bug, the same happens on a HP ProBook 440 G5.

ityogi, 2018-07-21:
This bug also affects HP Elitebook 840 G1.
Distribution: Debian 9
Kernel: 4.9.0-7-amd64
iio-sensor-proxy version 2.0-4

hadess, 2018-07-21:
This isn't a real problem. Load average is not a good representation of the CPU (or other resource) usage caused by an application. The driver requires us to poll for new information, which we do every couple of seconds.

closed 2018-07-21 by hadess

czirkoszoltan, 2018-07-25:
- You fill up the gas tank of your car
- Meter still shows its empty
- Take your car to a service center
- You're told that it's not a real problem, the car works

Yes, load average is not a good representation. However, the daemon shouldn't be in an uninterruptable sleep state. I don't know how the polling is implemented, but it is clearly wrong if doing something "every couple of seconds" causes the load to climb.

hadess, 2018-07-25:
[This](http://www.brendangregg.com/blog/2017-08-08/linux-load-averages.html) is how your gas tank works?
> However, the daemon shouldn't be in an uninterruptable sleep state.

It is every time it reads the sensor data, waiting on IO.

> I don't know how the polling is implemented, but it is clearly wrong if doing something "every couple of seconds" causes the load to climb.

I do know how the polling is implemented, and load isn't representative of the work that the daemon does (sleep 2 seconds, read a couple of data points).