hadess/iio-sensor-proxy #232: Lenovo Yoga S1 Laptop tablet buttons not working in tablet mode

includereserved, 2018-07-03:
This is not necessarily an issue with iio-sensor-proxy itself but is instead an issue with how the keyboard is disabled in tablet mode. (When I say tablet buttons I'm referring to the rotation lock and volume keys, perhaps the power button as well.)

hadess, 2018-07-03:
It's either a kernel driver problem, or a firmware problem. Most likely the latter I believe. Nothing we can do in iio-sensor-proxy, it doesn't make those changes.

closed 2018-07-03 by hadess

includereserved, 2018-07-03:
Yes I'm aware. I was hoping that because a lot of people using this are using Yogas there would be some discussion.

Anyway I wrote a small workaround in C that monitors /dev/input/by-path/platform-i8042-serio-0-event-kbd and reinterprets the presses since they're still getting that far.

hadess, 2018-07-03:
> Yes I'm aware. I was hoping that because a lot of people using this are using Yogas there would be some discussion.

I'm afraid that this isn't the right place to discuss this problem, because I'm the only one monitoring the issues here, and I can't fix your problem, as I don't have access to the hardware, and it's not a bug in iio-sensor-proxy.

> Anyway I wrote a small workaround in C that monitors /dev/input/by-path/platform-i8042-serio-0-event-kbd and reinterprets the presses since they're still getting that far.

If it gets to the OS, that means that the drivers are to blame, send a mail to ibm-acpi-devel@lists.sourceforge.net and platform-driver-x86@vger.kernel.org and they'll be able to answer your questions.