hadess/iio-sensor-proxy #233: Some error on installation

Migacz85, 2018-07-11:
Hi, I'm on manjaro trying to install it i got this errors:

./configure --prefix=/usr --sysconfdir=/etc
                                    
./configure: line 1930: syntax error near unexpected token `git-directory'
./configure: line 1930: `AX_IS_RELEASE(git-directory)'

some directions what next ?

hadess, 2018-07-13:
Install `autoconf-archive` from your distribution, or use your distribution's packages.

closed 2018-07-13 by hadess

Migacz85, 2018-07-13:
That strange because i have that package.

hadess, 2018-07-13:
You'll want to check the errors that might be there earlier in the output, rather than the last thing to fail.

Migacz85, 2018-07-13:
how ?

hadess, 2018-07-13:
> how ?

By reading the full output of the commands from top to bottom, rather than just the last few lines.

Migacz85, 2018-07-13:
That was my full output.

Migacz85, 2018-07-13:
> You needed to run autogen.sh before configure so that configure gets
> generated. There were errors in the autogen.sh run which you ignored.

Thank You very much.
following your steps i got results :

Making all in data
make[1]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/data'
  GEN      iio-sensor-proxy.service
make[1]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/data'
Making all in src
make[1]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/src'
  GEN      iio-sensor-proxy-resources.c
  GEN      iio-sensor-proxy-resources.h
make  all-am
make[2]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/src'
  CC       monitor_sensor-monitor-sensor.o
  CCLD     monitor-sensor
  CC       fake_input_accelerometer-fake-input-accelerometer.o
  CCLD     fake-input-accelerometer
  CC       test_mount_matrix-test-mount-matrix.o
  CC       test_mount_matrix-accel-mount-matrix.o
  CCLD     test-mount-matrix
  CC       test_orientation-test-orientation.o
  CC       test_orientation-orientation.o
  CC       test_orientation-accel-mount-matrix.o
  CCLD     test-orientation
  CC       test_orientation_gtk-test-orientation-gtk.o
  CC       test_orientation_gtk-orientation.o
  CCLD     test-orientation-gtk
  CC       iio_sensor_proxy-iio-sensor-proxy.o
  CC       iio_sensor_proxy-orientation.o
  CC       iio_sensor_proxy-drv-iio-buffer-accel.o
  CC       iio_sensor_proxy-drv-iio-poll-accel.o
  CC       iio_sensor_proxy-drv-input-accel.o
  CC       iio_sensor_proxy-drv-fake-compass.o
  CC       iio_sensor_proxy-drv-fake-light.o
  CC       iio_sensor_proxy-drv-iio-poll-light.o
  CC       iio_sensor_proxy-drv-hwmon-light.o
  CC       iio_sensor_proxy-drv-iio-buffer-light.o
  CC       iio_sensor_proxy-drv-iio-buffer-compass.o
  CC       iio_sensor_proxy-iio-buffer-utils.o
  CC       iio_sensor_proxy-accel-mount-matrix.o
  CC       iio_sensor_proxy-iio-sensor-proxy-resources.o
  CCLD     iio-sensor-proxy
make[2]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/src'
make[1]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/src'
Making all in docs
make[1]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/docs'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/docs'
make[1]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy'
make[1]: Nothing to be done for 'all-am'.
make[1]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy'

sudo make install

Making install in data
make[1]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/data'
make[2]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/data'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/etc/dbus-1/system.d'
 /bin/install -c -m 644 net.hadess.SensorProxy.conf '/etc/dbus-1/system.d'
 /bin/mkdir -p '/usr/lib/systemd/system'
 /bin/install -c -m 644 iio-sensor-proxy.service '/usr/lib/systemd/system'
 /bin/mkdir -p '/usr/lib/udev/rules.d'
 /bin/install -c -m 644 80-iio-sensor-proxy.rules '/usr/lib/udev/rules.d'
make[2]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/data'
make[1]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/data'
Making install in src
make[1]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/src'
make  install-am
make[2]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/src'
make[3]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/src'
 /bin/mkdir -p '/usr/bin'
  /bin/install -c monitor-sensor '/usr/bin'
 /bin/mkdir -p '/usr/sbin'
  /bin/install -c iio-sensor-proxy '/usr/sbin'
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/src'
make[2]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/src'
make[1]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/src'
Making install in docs
make[1]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/docs'
make[2]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy/docs'
make[2]: Nothing to be done for 'install-exec-am'.
Nothing to install
make[2]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/docs'
make[1]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy/docs'
make[1]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy'
make[2]: Entering directory '/home/migacz/Downloads/iio-sensor-proxy'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy'
make[1]: Leaving directory '/home/migacz/Downloads/iio-sensor-proxy'

Migacz85, 2018-07-13:
It looks that is working :)
Thank You very much. Any thoughts how i can induce this in Kde display ?

monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: right-up
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: right-up

Migacz85, 2018-07-13:
Thank You very much again, that project is a brilliant job !

hadess, 2018-07-13:
> Thank You very much. Any thoughts how i can induce this in Kde display ?

As per the README, no. I don't use or work on KDE.

Migacz85, 2018-07-13:
Once again thank You so much for help!
I figure it out, and using that simple script my screen is rotating
automatically :)

#!/bin/bash
monitor-sensor | while read line
do
if [[ $line =~ .*left.* ]]
then
    xrandr -o left
fi
if [[ $line =~ .*right.* ]]
then
    xrandr -o right
fi
if [[ $line =~ .*bottom-up.* ]]
then
    xrandr -o inverted
fi
if [[ $line =~ .*normal.* ]]
then
    xrandr -o normal
fi
done