hadess/iio-sensor-proxy #235: Suspend fails because proxy enters uninterruptible sleep indefinitely

chernetsov0, 2018-09-02:
This bug is closely related to the #231, but I couldn't care less about the load average - I just want my suspend to work. :smiley: 

The machine in question is **HP Envy 4-1257sr** and it happens to have the same **[ST LIS3LV02DL](https://www.st.com/resource/en/datasheet/lis3lv02dl.pdf)** accelerometer as in the ProBooks from #231.

```
P: /devices/platform/lis3lv02d/input/input8
E: ABS=7
E: DEVPATH=/devices/platform/lis3lv02d/input/input8
E: EV=9
E: ID_FOR_SEAT=input-platform-lis3lv02d
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_PATH=platform-lis3lv02d
E: ID_PATH_TAG=platform-lis3lv02d
E: IIO_SENSOR_PROXY_TYPE=input-accel
E: MODALIAS=input:b0019v0000p0000e0000-e0,3,kra0,1,2,mlsfw
E: NAME="ST LIS3LV02DL Accelerometer"
E: PHYS="lis3lv02d/input0"
E: PRODUCT=19/0/0/0
E: PROP=0
E: SUBSYSTEM=input
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:seat:systemd:
```

When I try to suspend, look what happens to the `/var/log/syslog`

```
Sep  2 03:07:35 hp-envy-4 systemd[1]: Reached target Sleep.
Sep  2 03:07:35 hp-envy-4 systemd[1]: Starting Suspend...
Sep  2 03:07:35 hp-envy-4 systemd-sleep[13858]: Suspending system...
Sep  2 03:07:35 hp-envy-4 kernel: [ 6388.121891] PM: suspend entry (deep)
Sep  2 03:07:55 hp-envy-4 kernel: [ 6388.121893] PM: Syncing filesystems ... done.
Sep  2 03:07:55 hp-envy-4 kernel: [ 6388.142616] Freezing user space processes ... 
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150305] Freezing of tasks failed after 20.007 seconds (1 tasks refusing to freeze, wq_busy=0):
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150336] iio-sensor-prox D    0   953      1 0x00000124
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150344] Call Trace:
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150360]  __schedule+0x291/0x8a0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150366]  schedule+0x2c/0x80
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150372]  schedule_timeout+0x15d/0x350
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150384]  ? __next_timer_interrupt+0xe0/0xe0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150389]  msleep+0x2d/0x40
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150398]  acpi_os_sleep+0xe/0x10
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150406]  acpi_ex_system_do_sleep+0x2c/0x36
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150413]  acpi_ex_opcode_1A_0T_0R+0x94/0xfd
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150422]  acpi_ds_exec_end_op+0x163/0x736
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150430]  acpi_ps_parse_loop+0x918/0x9c2
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150435]  ? acpi_ut_trace_ptr+0x2c/0x74
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150441]  acpi_ps_parse_aml+0x1ac/0x4bd
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150447]  acpi_ps_execute_method+0x1fa/0x2bc
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150452]  acpi_ns_evaluate+0x2ee/0x435
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150457]  acpi_evaluate_object+0x1d5/0x39f
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150467]  acpi_evaluate_integer+0x5d/0x100
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150474]  ? input_to_handler+0x97/0xf0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150482]  lis3lv02d_acpi_read+0x6f/0xa0 [hp_accel]
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150488]  ? input_pass_values.part.6+0x118/0x130
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150497]  lis3lv02d_read_8+0x29/0x50 [lis3lv02d]
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150503]  lis3lv02d_get_xyz+0x4c/0x1c0 [lis3lv02d]
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150511]  lis3lv02d_joystick_poll+0x4a/0xc0 [lis3lv02d]
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150519]  input_open_polled_device+0x35/0x50 [input_polldev]
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150524]  input_open_device+0x7c/0xa0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150531]  evdev_open+0x171/0x1c0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150539]  chrdev_open+0xc4/0x1b0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150547]  do_dentry_open+0x1c2/0x310
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150552]  ? __inode_permission+0x5b/0x160
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150558]  ? cdev_put.part.3+0x20/0x20
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150564]  vfs_open+0x4f/0x80
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150570]  path_openat+0x66e/0x1770
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150577]  ? compat_poll_select_copy_remaining+0x130/0x130
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150585]  do_filp_open+0x9b/0x110
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150593]  ? __check_object_size+0xaf/0x1b0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150600]  ? __alloc_fd+0x46/0x170
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150607]  do_sys_open+0x1bb/0x2c0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150612]  ? do_sys_open+0x1bb/0x2c0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150620]  SyS_openat+0x14/0x20
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150628]  do_syscall_64+0x73/0x130
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150635]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150639] RIP: 0033:0x7ff4ef732d19
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150642] RSP: 002b:00007fff342a6010 EFLAGS: 00000293 ORIG_RAX: 0000000000000101
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150648] RAX: ffffffffffffffda RBX: 000055f393529ad0 RCX: 00007ff4ef732d19
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150651] RDX: 0000000000080000 RSI: 000055f39355e980 RDI: 00000000ffffff9c
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150653] RBP: 000055f393559c60 R08: 0000000000000000 R09: 000055f3935493e0
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150656] R10: 0000000000000000 R11: 0000000000000293 R12: 0000000000000011
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.150659] R13: 000055f393520fc0 R14: 000055f393559c60 R15: 00007ff4f00c6280
Sep  2 03:07:55 hp-envy-4 gsd-power[1365]: Call to iio-proxy failed: Timeout was reached
Sep  2 03:07:55 hp-envy-4 gsd-power[2343]: Call to iio-proxy failed: Timeout was reached
Sep  2 03:07:55 hp-envy-4 kernel: [ 6408.360201] PM: suspend exit
```

To add insult to injury, I'm unable to `kill -9` the `iio-sensor-proxy` or to uninstall it via package manager, so it looks like it's in the uninterruptible sleep *indefinitely*. It has been a few hours since I issued `SIGKILL` and it's still there with the same PID.

This accelerometer is probably only used to park the HDD heads during drops and is pretty pointless on my machine (I did the SSD swap). But it is pretty generic, so it may be used in other HP notebooks for screen rotation as well.

I guess it's a hardware fault, so the accelerometer won't respond to `is3lv02d_acpi_read` forever, a situation which `iio-sensor-proxy` handles in a manner [Hachikō](https://en.wikipedia.org/wiki/Hachik%C5%8D) would be proud of. In this case it would probably be more graceful to timeout quite quickly or maybe blacklist the sensor for a while if it times out too many times in a row. Better loose autorotation in case of sensor failure then suspend/hibernate. And load average stays 2.0 while process is sleeping so maybe #231 had merit after all.

hadess, 2018-09-03:
I'm pretty sure that this is caused by the driver, though I wouldn't know, as I don't have that hardware at hand.

chernetsov0, 2018-09-03:
@hadess, any info I can provide to help? Also, is it possible to timeout such a call in any way or abort it? Or if the kernel blocks due to bad driver there is not much you can do?

hadess, 2018-09-04:
I can't do anything about this, please contact the kernel driver developers about this. The iio-sensor-proxy is trivial usage of the input API they provide.

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/235

closed 2020-01-09 by hadess