hadess/iio-sensor-proxy #237: Huawei Matebook X Pro Fails silently

rien333, 2018-09-19:
In contrast to https://github.com/hadess/iio-sensor-proxy/pull/234 running `iio-sensor-proxy` fails silently for me, although the fact that other matebook users have it working apparently (like in the aforementioned PR) leads me to believe that I'm doing something wrong. I'm running iio-sensor-proxy from the AUR git package on an updated Arch Linux. 

```shell
$ iio-sensor-proxy # no output
$ monitor-sensor &
Waiting for iio-sensor-proxy to appear
$ sudo iio-sensor-proxy # running as regular user achieves nothing
**
ERROR:monitor-sensor.c:165:appeared_cb: assertion failed (error == NULL): GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.10616 was not provided by any .service files (g-dbus-error-quark, 2)
fish: “monitor-sensor &” terminated by signal SIGABRT (Abort)
```
```shell 
$ udevadm info --export-db | grep --context=1 iio
P: /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/ACPI0008:00/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:00/ACPI0008:00/iio:device0
E: DEVTYPE=iio_device
E: MAJOR=242
E: MINOR=0
E: SUBSYSTEM=iio
```

hadess, 2018-09-19:
Please see the README about providing useful debugging information.

rien333, 2018-09-19:
I'm sorry, what kind of information am I missing?

rien333, 2018-09-19:
Weird that it is complaining about not finding sensors while others report it working on the same machine.

hadess, 2018-09-19:
From the README:
> If that doesn't work, please file an issue, make sure any running iio-sensor-proxy has been stopped: `systemctl stop iio-sensor-proxy.service` and attach the output of: `G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy` running as root.

> Weird that it is complaining about not finding sensors while others report it working on the same machine.

You, or the person compiling the kernel you're using, probably didn't enable the drivers necessary for it.

rien333, 2018-09-19:
I'm running arch linux with a stock and stable kernel (`4.18.8-arch1-1-ARCH`), but thanks for pointing me in the right direction.

rien333, 2018-09-19:
Oddly enough `/sys/bus/iio/devices/iio:device0/in_illuminance_raw` seems to work fine, so is it really a matter of something not being enabled in the kernel?

hadess, 2018-10-09:
We don't use `in_illuminance_raw` at all. It's not read in the driver, and it's not a file we use in the udev rules. Maybe a newer kernel has support for the `in_illuminance_input` file instead? Or maybe it regressed.

I don't know what iio device it uses, so I can't look for changes in the kernel.

rien333, 2018-10-09:
I have quite a new kernel (just plain arch) so idk. Not totally sure what all those files are, but `/sys/bus/iio/devices/iio:device0/in_illuminance_input` reports the same value as `in_illuminance_raw`. Is the former one the one used by this project?

hadess, 2018-10-09:
If both exist, then it should work fine (and iio-sensor-proxy would use `in_illuminance_input`)

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/237

closed 2020-01-09 by hadess