hadess/iio-sensor-proxy #244: Should monitor-sensor reflect orientation adjusted by udev hwdb?

iam-TJ, 2018-10-29:
I'm trying to fix the rotation of an Asus T300CHI 2-in-1 transformer. I'm making changes to the udev hwdb and they are reflected in a query of the db but "monitor-sensor" still reports the same orientation as it does without the hwdb change. Should it be reporting the orientation as changed by hwdb?
```
$ cat /sys/class/dmi/id/modalias
dmi:bvnAmericanMegatrendsInc.:bvrT300CHI.207:bd08/06/2015:svnASUSTeKCOMPUTERINC.:pnT300CHI:pvr1.0:rvnASUSTeKCOMPUTERINC.:rnT300CHI:rvr1.0:cvnASUSTeKCOMPUTERINC.:ct10:cvr1.0:

$ cat /sys/`udevadm info -q path -n /dev/iio:device1`/../modalias
acpi:INVN6500:PNP0C50:

$cat /etc/udev/hwdb.d/61-sensor-local.hwdb 
# original: /sys/class/i2c-adapter/i2c-0/i2c-INVN6500\:00/iio\:device1/in_accel_mount_matrix = 1, 0, 0; 0, 1, 0; 0, 0, 1
sensor:modalias:acpi:INVN6500*:dmi:*svnASUSTeK*:pnT300CHI*
 ACCEL_MOUNT_MATRIX=0, 1, 0; -1, 0, 0; 0, 0, 1

$ systemd-hwdb query sensor:modalias:acpi:INVN6500*:dmi:*svnASUSTeK*:pnT300CHI*
ACCEL_MOUNT_MATRIX=0, 1, 0; -1, 0, 0; 0, 0, 1

$ monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: right-up)
```
This is whilst using classic laptop orientation so it is reporting "right-up" when it should be "normal", which if I've got it correct, means the adjustment should be +90 degrees (clockwise).

iam-TJ, 2018-10-29:
I found I had to restart iio-sensor-proxy to have monitor-sensor see the changes, despite the fact that iio-sensor-proxy was logging a change to the udev database.

And the correct working setting I found to be:
```
$ cat /etc/udev/hwdb.d/61-sensor-local.hwdb 
# /sys/class/i2c-adapter/i2c-0/i2c-INVN6500\:00/iio\:device1/in_accel_mount_matrix = 1, 0, 0; 0, 1, 0; 0, 0, 1
sensor:modalias:acpi:INVN6500*:dmi:*svnASUSTeK*:pnT300CHI*
#                  swap L<>R; rotate +90; no change
 ACCEL_MOUNT_MATRIX=0, -1, 0; 1, 0, 0; 0, 0, 1
```

hadess, 2018-10-29:
> I found I had to restart iio-sensor-proxy to have monitor-sensor see the changes, despite the fact that iio-sensor-proxy was logging a change to the udev database.

We only read the mount matrix on startup, when the device is initialised. Adding a helper to monitor the mount matrix might be a good idea though, even if the use case is rather limited.

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/244

closed 2020-01-09 by hadess