hadess/iio-sensor-proxy #247: Started IIO Sensor Proxy Service ... Deepin OS boot freezes on my laptop(HP 2570p)

aiamuzz, 2018-11-22:
@hadess ...

I am sorry ... not sure if my posting this issue is right or wrong ... but the phrase ... **Started IIO Sensor Proxy Service** ... is a nightmarish sight to me ... this phrase on my screen mean uncertainty of being able to boot successfully on my laptop.

this is the gory nightmarish screen that gives me sleepless nights ...
[image]

I have a HP 2570p laptop ... I run Deepin OS on it ... as a result of this nightmarish encounter of the above phrase on my screen ... and with the hope that i can evade this phrase popping on my screen ... i have gone to the extent of installing the same Deepin OS on 3 separate partitions alongside my Windows 10 OS.

I tried to google this term and was led to your github repo ... hoping you'd be able to clarify/help me get rid of this ...

1. Is this package essential to my OS(Deepin) ?
2. I haven't yet tried it but will uninstalling it cause any other issues. Clearly as per your repo page this is a package for accelerometer ... and clearly my laptop is non-touch based laptop 
3. Would you want me to run some diagnostic commands on my system to ascertain if this package is indeed causing my nightmares ?

some observations for your perusal ...

Almost all(more than 90%) boot issues happen to freeze up at the above screen ... its my version of BSOD ... Ctrl+Alt+Del ... is the only way to get out of this hang ... today i had a bout of 11 unsuccessful boot freezes ... and almost all ended up at this screen.

I have been looking through the journald logs ... and i find that the boot logs of a successful as this sensor service missing all together at times ... and present at certain other times ... but the boot freezes are almost all the time end up at the above screen.

I am following up with Deepin OS team as well on this issue(even they are scratching their heads currently) ... moreover seems i am the only one affected ... may be my specific hardware is allergic to this boot freezes ...

Hope i could find some answers at least here

hadess, 2018-11-22:
The problem is likely not caused by iio-sensor-proxy in any case. As you can see, it started successfully.

Check with your distribution on how to debug this.

(By the way, most HP laptops have an accelerometer, it's used to park the hard disk in case of a fall.)

closed 2018-11-22 by hadess

aiamuzz, 2018-11-22:
> The problem is likely not caused by iio-sensor-proxy in any case. As you can see, it started successfully.
> 
> Check with your distribution on how to debug this.
> 
> (By the way, most HP laptops have an accelerometer, it's used to park the hard disk in case of a fall.)

Oh ... didn't know that HP laptops had an accelerometer ... ok ... thanks for looking !