hadess/iio-sensor-proxy #252: Wrong orientations by monitor-sensor on Medion Akoya E2212T

Tuxman2, 2019-01-09:
Hello,

I've got a big problem with iio-proxy-service and the monitor-sensor command. Monitor-sensor command don't give me the right orientations. That's why I posted this message here.

Here are the results in my case:

First orientation found by monitor-sensor command : Normal (it's good).
Screen tilt to the right -> Orientation = right-up (it's wrong because it must be left-up)
Screen tilt to landscape -> Orientation = normal (It's good)
Screen tilt to the left -> Orientation = left-up (it's wrong because it must be right-up)
Screen tilt to landscape -> Orientation = normal (always good)
Screen tilt to 360° -> Orientation = bottom-up (it's good)

So there is a bad orientation between left and right (inverted).

I created the /etc/udev/hwdb.d/61-sensor-local.hwdb and put the following information:

sensor:modalias:acpi:SMO8500*:dmi::svnMEDION:pnWCBT112X:
ACCEL_MOUNT_MATRIX=0, -1, 0; 1, 0, 0; 0, 0, 0

and use that command: sudo systemd-hwdb update

The results : Nothing change.

I don't know how to solve this problem. That's why I ask your help.

Note: I use the systemd 237-3ubuntu10.9 version, and my desktop environment is Cinnamon 4.0.9

Your test-orientation-gtk program give me these values:

0,-1,0 -> normal
-1,0,0 -> right-up
1,0,0 -> left-up
0,1,0 -> bottom-up

Thanks.

hadess, 2019-01-09:
> sensor:modalias:acpi:SMO8500*:dmi::svnMEDION:pnWCBT112X:
> ACCEL_MOUNT_MATRIX=0, -1, 0; 1, 0, 0; 0, 0, 0
> 
> and use that command: sudo systemd-hwdb update
> 
> The results : Nothing change.

Did you verify that the quirk gets assigned correctly in `udevadm info --export-db`?

FWIW, this mount matrix is incorrect so won't be applied anyway, you have no Z axis. I can't help you find the right values for it, but it needs to have all 3 axis available.

> 0,-1,0 -> normal
> -1,0,0 -> right-up
> 1,0,0 -> left-up
> 0,1,0 -> bottom-up

Values are between -256 and 256.

Seeing as it's not an iio-sensor-proxy bug, and you've already got a systemd issue open for the quirk, I'll close this.

closed 2019-01-09 by hadess

spikerguy, 2019-10-29:
I have similar issue, Linx 10 Tablet testing wayfire, but this guide giving 404 error.
See 60-sensor.hwdb for details. Is anyone still working on this ?