hadess/iio-sensor-proxy #256: My Jumper EZpad 6 Pro has a different modalias

Zennoe, 2019-03-02:
In line 293 of the hwdb file it tries to match the same tablet but the revision number within the string is different.

Line 293: `sensor:modalias:acpi:BOSC0200*:dmi:*:svnJumper:pnEZpad:*:rvr.A006:*`

My actual modalias: `dmi:bvnJumper.A116:bvr5.12:bd04/24/2018:svnJumper:pnEZpad:pvr:rvnJumper:rnEZpad:rvr.A009:cvn:ct10:cvr:`

Zennoe, 2019-03-02:
After some testing it turned out that the accel matrix is a bit different. Apparently there is an EZpad 6 Pro and an EZpad 6s Pro. This is the latter.

The correct matrix for the *EZpad 6s Pro* is: `ACCEL_MOUNT_MATRIX=1, 0, 0; 0, -1, 0; 0, 0, 1`

hadess, 2019-03-02:
The hwdb file is part of systemd, not iio-sensor-proxy.

closed 2019-03-02 by hadess