hadess/iio-sensor-proxy #258: Calibration for Asus T500LAB

Phulkrum, 2019-03-20:
Finished building from source on Ubuntu 18.4 after removing version 2.4-2 which is supplied by Ubuntu repo. both versions (2.4-2 and 2.5 built from source) exhibit the same result, system boots and once logged in screen defaults to right-up rotation.  leaving me to have to physically turn the laptop to that position then lock the rotation to be able to use it in laptop mode. is there a way to modify the service on my side to default to "normal"? (calibrate for my system's accelerometer? )

hadess, 2019-03-20:
> is there a way to modify the service on my side to default to "normal"?

Yes, and it's even explained in the README:
https://github.com/hadess/iio-sensor-proxy#accelerometer-orientation

The quirks live in systemd, and this is probably the quirk for the device that's closest to yours:
https://github.com/systemd/systemd/blob/master/hwdb/60-sensor.hwdb#L119

closed 2019-03-20 by hadess

Phulkrum, 2019-03-21:
Thanks, 
found this after further digging Much appreciated and should have dug before bugging...