hadess/iio-sensor-proxy #259: Sensors broken (regression) on Yoga 2 Pro

dos1, 2019-03-23:
On my Lenovo Yoga 2 Pro, commit ccaf37e00a53047762b16a6fa8929fae9d741145 breaks accelerometer detection, while commit 79c79270b92bdd01e4aaf3d8d63089b02f42e312 additionally breaks the light sensor and commit 924dd58464f0862631d53cbcb971cfc237c309a7 breaks the compass as well.

Afterwards, the only thing I get from iio-sensor-proxy is:
```
** (process:22489): DEBUG: 01:50:28.475: Could not find any supported sensors
```

`udevadm info --export-db`:
```
P: /devices/pci0000:00/0000:00:14.0/usb2/2-6
N: bus/usb/002/003
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6
E: SUBSYSTEM=usb
E: DEVNAME=/dev/bus/usb/002/003
E: DEVTYPE=usb_device
E: DRIVER=usb
E: PRODUCT=2047/855/1151
E: TYPE=0/0/0
E: BUSNUM=002
E: DEVNUM=003
E: MAJOR=189
E: MINOR=130
E: USEC_INITIALIZED=2434034
E: ID_VENDOR=Invensense
E: ID_VENDOR_ENC=Invensense
E: ID_VENDOR_ID=2047
E: ID_MODEL=Lenovo_Yoga
E: ID_MODEL_ENC=Lenovo\x20Yoga\x20\x20\x20\x20\x20\x20\x20\x20\x20
E: ID_MODEL_ID=0855
E: ID_REVISION=1151
E: ID_SERIAL=Invensense_Lenovo_Yoga_F9FB836F29002600
E: ID_SERIAL_SHORT=F9FB836F29002600
E: ID_BUS=usb
E: ID_USB_INTERFACES=:030000:
E: ID_VENDOR_FROM_DATABASE=Texas Instruments
E: ID_MODEL_FROM_DATABASE=Invensense Embedded MotionApp HID Sensor

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0
E: SUBSYSTEM=usb
E: DEVTYPE=usb_interface
E: DRIVER=usbhid
E: PRODUCT=2047/855/1151
E: TYPE=0/0/0
E: INTERFACE=3/0/0
E: MODALIAS=usb:v2047p0855d1151dc00dsc00dp00ic03isc00ip00in00
E: USEC_INITIALIZED=2437277
E: ID_VENDOR_FROM_DATABASE=Texas Instruments
E: ID_MODEL_FROM_DATABASE=Invensense Embedded MotionApp HID Sensor

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001
E: SUBSYSTEM=hid
E: DRIVER=hid-sensor-hub
E: HID_ID=0003:00002047:00000855
E: HID_NAME=Invensense Lenovo Yoga         
E: HID_PHYS=usb-0000:00:14.0-6/input0
E: HID_UNIQ=F9FB836F29002600
E: MODALIAS=hid:b0003g0003v00002047p00000855

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200001.1.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200001.1.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: MODALIAS=platform:HID-SENSOR-200001

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: DRIVER=hid_sensor_als
E: MODALIAS=platform:HID-SENSOR-200041

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/iio:device0
N: iio:device0
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/iio:device0
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device0
E: DEVTYPE=iio_device
E: MAJOR=237
E: MINOR=0

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/trigger0
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/trigger0
E: SUBSYSTEM=iio

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: DRIVER=hid_sensor_accel_3d
E: MODALIAS=platform:HID-SENSOR-200073

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/iio:device1
N: iio:device1
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/iio:device1
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device1
E: DEVTYPE=iio_device
E: MAJOR=237
E: MINOR=1

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/trigger1
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/trigger1
E: SUBSYSTEM=iio

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200076.3.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200076.3.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: DRIVER=hid_sensor_gyro_3d
E: MODALIAS=platform:HID-SENSOR-200076

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200076.3.auto/iio:device3
N: iio:device3
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200076.3.auto/iio:device3
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device3
E: DEVTYPE=iio_device
E: MAJOR=237
E: MINOR=3

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200076.3.auto/trigger3
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200076.3.auto/trigger3
E: SUBSYSTEM=iio

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: DRIVER=hid_sensor_magn_3d
E: MODALIAS=platform:HID-SENSOR-200083

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/iio:device2
N: iio:device2
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/iio:device2
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device2
E: DEVTYPE=iio_device
E: MAJOR=237
E: MINOR=2

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/trigger2
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/trigger2
E: SUBSYSTEM=iio

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200086.5.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200086.5.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: DRIVER=hid_sensor_incl_3d
E: MODALIAS=platform:HID-SENSOR-200086

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200086.5.auto/iio:device5
N: iio:device5
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200086.5.auto/iio:device5
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device5
E: DEVTYPE=iio_device
E: MAJOR=237
E: MINOR=5

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200086.5.auto/trigger5
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200086.5.auto/trigger5
E: SUBSYSTEM=iio

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-20008a.6.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-20008a.6.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: DRIVER=hid_sensor_rotation
E: MODALIAS=platform:HID-SENSOR-20008a

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-20008a.6.auto/iio:device4
N: iio:device4
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-20008a.6.auto/iio:device4
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device4
E: DEVTYPE=iio_device
E: MAJOR=237
E: MINOR=4

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-20008a.6.auto/trigger4
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-20008a.6.auto/trigger4
E: SUBSYSTEM=iio

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-2000e1.7.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-2000e1.7.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: MODALIAS=platform:HID-SENSOR-2000e1

P: /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-2000e2.8.auto
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-2000e2.8.auto
E: SUBSYSTEM=platform
E: DEVTYPE=mfd_device
E: MODALIAS=platform:HID-SENSOR-2000e2
```

hadess, 2019-03-23:
Where did you install iio-sensor-proxy?

dos1, 2019-03-23:
At first I have installed the Arch package, then started bisecting it from the git checkout.

hadess, 2019-03-23:
What's the output of `ls -lR /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto` ?

Looks like the "name is accel_3d" and checks in https://github.com/hadess/iio-sensor-proxy/blob/master/data/80-iio-sensor-proxy.rules don't match the same devices.

hadess, 2019-03-23:
Also make sure that the rules file is installed in the correct location for your system, and actually processed by udev.

dos1, 2019-03-23:
Of course that was it - the file was installed, but not processed yet. Sorry for the noise :)

closed 2019-03-23 by dos1