hadess/iio-sensor-proxy #261: Disabling the ambient light change on the Lenovo Yoga 720-15IKB

Circl3s, 2019-04-10:
So, as the title suggests, iio-sensor-proxy works flawlessly for me on the Lenovo Yoga 720-15IKB running Manjaro Budgie on the 5.0.5-1 kernel.
```
~ >>> monitor-sensor                                                                               
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: normal)
=== Has ambient light sensor (value: 44.000000, unit: lux)
    Light changed: 33.000000 (lux)
    Light changed: 44.000000 (lux)
    Light changed: 55.000000 (lux)
    Light changed: 44.000000 (lux)
    Light changed: 55.000000 (lux)
    Accelerometer orientation changed: left-up
    Light changed: 44.000000 (lux)
    Light changed: 33.000000 (lux)
    Light changed: 44.000000 (lux)
    Light changed: 33.000000 (lux)
    Accelerometer orientation changed: normal
    Light changed: 44.000000 (lux)
    Light changed: 55.000000 (lux)
    Light changed: 44.000000 (lux)
    Light changed: 55.000000 (lux)
    Light changed: 66.000000 (lux)
    Light changed: 55.000000 (lux)
    Light changed: 44.000000 (lux)
...
```
However, the ambient light sensor is really sensitive. My brightness keeps changing a bit, like, 2 times a second even if nothing is happening (for example the screen gets darker every time I start typing on the keyboard because my hands cast shadow on the sensor). Is there a way to disable this?

hadess, 2019-04-10:
If nothing consumes the data from the sensor, then it won't be read. Disabling "Automatic Brightness" functionality in GNOME should do that, for example.

closed 2019-04-10 by hadess

Circl3s, 2019-04-10:
Wow, completely missed that in the settings, thanks!