hadess/iio-sensor-proxy #266: Lenovo Yoga 11-ISK accelerometer orientation changes constantly

selankon, 2019-04-29:
Hi

I have a Lenovo Yoga 11-ISK  with a linux mint 19.1. If I execute `monitor-sensor` I get an orientation change constantly, here a sample output:

```
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: right-up
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: right-up
    Accelerometer orientation changed: normal
    Accelerometer orientation changed: right-up
```
I can't find information about how to calibrate it or what is the cause of the problem. Any ideas?

hadess, 2019-05-14:
Please test with version 2.6.

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/266

closed 2020-01-09 by hadess