hadess/iio-sensor-proxy #273: Orientation: Undefined on X1 Yoga (G3) Antergos (Arch based)

edhilgendorf, 2019-05-20:
system
------------------------------------------------------------------
5.1.2-arch1-1-ARCH (Antergos) on X1 Yoga (G3)

monitor-sensor
------------------------------------------------------------------
   Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)

iio-sensor-proxy status
------------------------------------------------------------------
● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/usr/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: disabled)
   Active: active (running) since Sun 2019-05-19 15:26:15 PDT; 1h 35min ago
 Main PID: 680 (iio-sensor-prox)
    Tasks: 3 (limit: 4915)
   Memory: 1.9M
   CGroup: /system.slice/iio-sensor-proxy.service
           └─680 /usr/bin/iio-sensor-proxy

May 19 15:26:15 fireside systemd[1]: Starting IIO Sensor Proxy service...
May 19 15:26:15 fireside systemd[1]: Started IIO Sensor Proxy service.
May 19 15:29:04 fireside systemd[1]: iio-sensor-proxy.service: Current command vanished from the unit file, execution of the command list won't be resumed.

edhilgendorf, 2019-05-20:
Debug
`** (process:29051): DEBUG: 17:07:50.073: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0001/HID-SENSOR-200041.3.auto/iio:device1
** (process:29051): DEBUG: 17:07:50.075: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0001/HID-SENSOR-200041.3.auto/iio:device1 of type ambient light sensor at IIO Buffer Light sensor
** (process:29051): DEBUG: 17:07:50.076: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/trigger0
** (process:29051): DEBUG: 17:07:50.076: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0
** (process:29051): DEBUG: 17:07:50.076: No autodetected location, falling back to display location
** (process:29051): DEBUG: 17:07:50.076: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:29051): DEBUG: 17:07:50.081: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/trigger0
** (process:29051): DEBUG: 17:07:50.082: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0/scan_elements/in_accel_x_en
** (process:29051): DEBUG: 17:07:50.082: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0/scan_elements/in_accel_z_en
** (process:29051): DEBUG: 17:07:50.082: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0/scan_elements/in_timestamp_en
** (process:29051): DEBUG: 17:07:50.082: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0/scan_elements/in_accel_y_en
** (process:29051): DEBUG: 17:07:50.082: Trying to read 'in_accel_x_scale' (name) from dir '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0'
** (process:29051): DEBUG: 17:07:50.083: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0/in_accel_x_scale: No such file or directory
** (process:29051): DEBUG: 17:07:50.083: Trying to read 'in_accel_scale' (generic name) from dir '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0'
** (process:29051): DEBUG: 17:07:50.083: Trying to read 'in_accel_x_offset' (name) from dir '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0'
** (process:29051): DEBUG: 17:07:50.083: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0/in_accel_x_offset: No such file or directory
** (process:29051): DEBUG: 17:07:50.083: Trying to read 'in_accel_offset' (generic name) from dir '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0'
** (process:29051): DEBUG: 17:07:50.083: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:29051): DEBUG: 17:07:50.083: Trying to read 'in_accel_z_scale' (name) from dir '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0'
** (process:29051): DEBUG: 17:07:50.083: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0/in_accel_z_scale: No such file or directory
** (process:29051): DEBUG: 17:07:50.083: Trying to read 'in_accel_scale' (generic name) from dir '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0'
** (process:29051): DEBUG: 17:07:50.083: Trying to read 'in_accel_z_offset' (name) from dir '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0'
** (process:29051): DEBUG: 17:07:50.083: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0/in_accel_z_offset: No such file or directory
** (process:29051): DEBUG: 17:07:50.083: Trying to read 'in_accel_offset' (generic name) from dir '/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device0'
** (process:29051): DEBUG: 17:07:50.083: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
[... cut ...]
** (process:29051): DEBUG: 17:07:54.996: process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4 bytes: 4 is_signed: 1 be: 0 shift: 0 bits_used: 32
** (process:29051): DEBUG: 17:07:54.996: process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8 bytes: 4 is_signed: 1 be: 0 shift: 0 bits_used: 32
** (process:29051): DEBUG: 17:07:54.996: Accel read from IIO on 'iio:device0': 0, 0, 0 (scale 0.000010)
** (process:29051): DEBUG: 17:07:54.996: Accel sent by driver (quirk applied): 0, 0, 0 (scale: 0.000010)`

edhilgendorf, 2019-05-20:
iio-sensor-proxy-2.6-1

hadess, 2019-05-20:
Duplicate of #270

closed 2019-05-20 by hadess