hadess/iio-sensor-proxy #274: iio-sensor-proxy.service: Main process exited, code=killed, status=15/TERM

gardeaz2, 2019-05-23:
Hello,

My iio-sensor-proxy doesn't seem to be working, and I am not sure why. When I run 
`[daniel@daniel-pc ~]$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy`,
I get the message
`Error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files`

After noticing this, I restarted it and checked the status,
`[daniel@daniel-pc ~]$ systemctl restart iio-sensor-proxy.service`
`[daniel@daniel-pc ~]$ systemctl status iio-sensor-proxy.service`,
which returned
`iio-sensor-proxy.service - IIO Sensor Proxy service`
`   Loaded: loaded (/usr/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: disabled)`
`   Active: inactive (dead)`

`May 23 15:28:56 daniel-pc systemd[1]: Starting IIO Sensor Proxy service...`
`May 23 15:28:56 daniel-pc systemd[1]: Started IIO Sensor Proxy service.`
`May 23 15:28:56 daniel-pc systemd[1]: iio-sensor-proxy.service: Main process exited, code=killed, status=15/TERM`
`May 23 15:28:56 daniel-pc systemd[1]: iio-sensor-proxy.service: Succeeded.`

The same message appears if I use `systemctl start iio-sensor-proxy.service` instead of `systemctl restart iio-sensor-proxy.service`.

I'm not sure how to proceed from here. What does status=15/term mean? What else can I do to try to solve my issue?

Thanks, and let me know if you need any more information.

hadess, 2019-07-10:
How did you install iio-sensor-proxy, and which version?

gardeaz2, 2019-07-11:
I installed using Git to download the files, then ran 
`./configure --prefix=/usr --sysconfdir=/etc`
`make`
`make install`
as the instructions said on the readme.

I also tried installing it from Pamac,, but I got the same error.

Both installations use version 2.7.

I'm running Manjaro KDE, if that is relevant.

cheeseypi, 2019-09-08:
I am having this issue as well. I installed from Pacman on arch, then uninstalled and attempted install from repo, but the compilation failed.

hadess, 2019-09-09:
Can you please follow the instructions in the README about debugging, and get the debug output? Have you also tried running iio-sensor-proxy on its own with that environment variable, as root, to see if there were any errors?

0xdc, 2019-11-15:
Ran into the same issue on clearlinux with git master. I rebooted to get the udev rules.

```
$ udevadm info --export-db | grep iio
P: /devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0001/HID-SENSOR-200041.5.auto/iio:device0
N: iio:device0
E: DEVPATH=/devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0001/HID-SENSOR-200041.5.auto/iio:device0
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device0
E: DEVTYPE=iio_device
E: IIO_SENSOR_PROXY_TYPE=iio-buffer-als
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: SUBSYSTEM=iio
P: /devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device3
N: iio:device3
E: DEVPATH=/devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200073.14.auto/iio:device3
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device3
E: DEVTYPE=iio_device
E: IIO_SENSOR_PROXY_TYPE=iio-buffer-accel
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: SUBSYSTEM=iio
P: /devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200076.12.auto/iio:device2
N: iio:device2
E: DEVPATH=/devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0002/HID-SENSOR-200076.12.auto/iio:device2
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device2
E: DEVTYPE=iio_device
E: SUBSYSTEM=iio
...
$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
Error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files
```

```
# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:1752): DEBUG: 07:14:15.356: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.
```

hadess, 2019-11-15:
`iio-sensor-proxy is already running, or it cannot own its D-Bus name.`
coupled with
`The name net.hadess.SensorProxy was not provided by any .service files`
means D-Bus couldn't find the files you just installed. Install them in the right prefix (and maybe reboot). This isn't an iio-sensor-proxy problem. Maybe using the version that comes with your distribution would be a good idea as well.

0xdc, 2019-11-15:
Ah you're right! The dbus wasn't picking up the file in /etc/dbus-1/system.d. I moved it to /usr/share/dbus-1/system.d and it starts now.

hadess, 2020-01-08:
I'm going to assume that the original poster had the same problem that was just diagnosed.

closed 2020-01-08 by hadess