hadess/iio-sensor-proxy #275: HP Envy x360 - no iio sensors detected

constantbeginner, 2019-06-08:
Hello there,

on my HP Envy x360 with Ubuntu 19.04 the command `udevadm info --export-db` does not detect any iio device.

`lsmod | grep acc` though finds this:
```
hp_accel               28672  0
lis3lv02d              24576  1 hp_accel
```
I have tried using `sudo systemctl start iio-sensor-proxy` which executes without giving any output whatsoever, so does `iio-sensor-proxy` 

`gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy `produces the following error message:

`Error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files`

launching `monitor-sensor` only produces `Waiting for iio-sensor-proxy to appear`, which is plausible, since no sensor is detected.

What am I missing?

linusw, 2019-06-13:
This seems to be using the old lis3lv02d sensor driver from drivers/misc/lis3lv02d/

I don't think that is a good idea. It uses a custom chardev and not the IIO subsystem, is not very well maintained and there is a modern driver for it. In my opinon the kernel need to be reconfigured not to build this module and use drivers/iio/accel/st_accel_i2c.c instead.

Possibly the IIO driver needs to be patched to properly detect the lis3lv02d on x86/ACPI as well.

constantbeginner, 2019-06-15:
Unfortunately this is all way above my skill level. If there is anything I could do to help, I would be happy to try. Would that be useful if I _try_ to reconfigure the kernel (never done it before) and let you know the outcome?

hadess, 2019-07-10:
> I don't think that is a good idea. It uses a custom chardev and not the IIO subsystem, is not very well maintained and there is a modern driver for it. In my opinon the kernel need to be [...]

`lis3lv02d_joystick_enable()` in `drivers/misc/lis3lv02d/lis3lv02d.c` will export the accelerometer data through an input device, which iio-sensor-proxy supports (`src/drv-input-accel.c`).

You'll want to check whether the kernel is correctly configured to export that input device, and whether you have a recent version of iio-sensor-proxy.

See also !173.

wismill, 2019-07-25:
@constantbeginner  There seem to be a [fix in preparation](https://bugzilla.kernel.org/show_bug.cgi?id=199715#c10).

constantbeginner, 2019-07-25:
@wismill That's great news, I'll sit tight and wait for it to be released, then I'll be happy to beta test it and let you know

wismill, 2019-09-28:
Fix is still [in preparation](https://lists.freedesktop.org/archives/amd-gfx/2019-September/040553.html).

constantbeginner, 2019-09-28:
I'm still available to test it once it's out

hadess, 2019-09-29:
I'll close this seeing as the reason why there's no accelerometer is because none are accessible by the kernel. Please file a new bug once the driver's landed and if there's problems.

closed 2019-09-29 by hadess

wismill, 2020-01-11:
https://www.phoronix.com/scan.php?page=news_item&px=AMD-Sensor-Fusion-Hub-Linux