hadess/iio-sensor-proxy #277: accelerometer in toshiba z20t

dragonlost, 2019-06-22:
Hello. 

I am testing ubuntu on my new laptop. Everything works very well and the automatic brightness works too.

On the other hand, the automatic rotation does not work.
Yet with monitor-sensor we see very well the change of position of the accelerometer.

I am under Ubuntu Gnome 18.04.2 Amd64 with version 2.4-2 of iio-sensor-proxy.

I also have a small bug on the brightness, indeed the automatic adjustment tends to go too much beats. Indeed the last level of brightness cuts the backlight of my screen.

dragonlost, 2019-06-22:
log for `G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy` command in root : 
[test.txt](https://github.com/hadess/iio-sensor-proxy/files/3317062/test.txt)

hadess, 2019-06-25:
The logs show that the orientation is correctly computed. If rotation doesn't work automatically it might be because you disabled it, or because there's a bug in the version of the desktop you're using. You should report the problem to your distribution.

closed 2019-06-25 by hadess