hadess/iio-sensor-proxy #280: Hard drive accelerator used as input

janbrummer, 2019-09-01:
On a HP EliteBook 850 G3 the hard driver accelerator sensor  `lis3lv02d ` is accidently used as an input source for iio-sensor-proxy. This causes lots of problems taking this notebook to a meeting in an active state. Screen orientation changes quickly and it looks akwards in a meeting room rotating the whole system to get back to a correct state.
Input device is created by `hp_accel` module: `Glue between LIS3LV02Dx and HP ACPI BIOS and support for disk protection LED.`

How can we prevent it from getting used?

The necessary information for the sensor:

```
P: /devices/platform/lis3lv02d
L: 0
E: DEVPATH=/devices/platform/lis3lv02d
E: SUBSYSTEM=platform
E: MODALIAS=platform:lis3lv02d
E: USEC_INITIALIZED=32677139
E: ID_PATH=platform-lis3lv02d
E: ID_PATH_TAG=platform-lis3lv02d

P: /devices/platform/lis3lv02d/input/input24
L: 0
E: DEVPATH=/devices/platform/lis3lv02d/input/input24
E: SUBSYSTEM=input
E: PRODUCT=19/0/0/0
E: NAME="ST LIS3LV02DL Accelerometer"
E: PHYS="lis3lv02d/input0"
E: PROP=0
E: EV=9
E: ABS=7
E: MODALIAS=input:b0019v0000p0000e0000-e0,3,kra0,1,2,mlsfw
E: USEC_INITIALIZED=33141993
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_PATH=platform-lis3lv02d
E: ID_PATH_TAG=platform-lis3lv02d
E: ID_FOR_SEAT=input-platform-lis3lv02d
E: IIO_SENSOR_PROXY_TYPE=input-accel
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:seat:

P: /devices/platform/lis3lv02d/input/input24/event18
N: input/event18
L: 0
S: input/by-path/platform-lis3lv02d-event
E: DEVPATH=/devices/platform/lis3lv02d/input/input24/event18
E: SUBSYSTEM=input
E: DEVNAME=/dev/input/event18
E: MAJOR=13
E: MINOR=82
E: USEC_INITIALIZED=33932564
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_PATH=platform-lis3lv02d
E: ID_PATH_TAG=platform-lis3lv02d
E: IIO_SENSOR_PROXY_TYPE=input-accel
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: LIBINPUT_DEVICE_GROUP=19/0/0:lis3lv02d
E: DEVLINKS=/dev/input/by-path/platform-lis3lv02d-event
E: TAGS=:systemd:

P: /devices/platform/lis3lv02d/input/input24/js0
N: input/js0
L: 0
E: DEVPATH=/devices/platform/lis3lv02d/input/input24/js0
E: SUBSYSTEM=input
E: DEVNAME=/dev/input/js0
E: MAJOR=13
E: MINOR=0
E: USEC_INITIALIZED=33147579
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_PATH=platform-lis3lv02d
E: ID_PATH_TAG=platform-lis3lv02d
E: IIO_SENSOR_PROXY_TYPE=input-accel
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
```

hadess, 2019-09-02:
Disabling the driver should do it.

janbrummer, 2019-09-02:
Yeah, right but then i loose disk protection. Not a good deal.

hadess, 2019-09-02:
You still have a spinning disk in this machine? As far as I'm concerned, if the `hp_accel` driver creates an accelerometer device, then it needs to be consumable by user-space, otherwise it could all be handled internally.

I thought that the main problem with the HP devices were the orientation of the accelerometer.

Try setting `ID_INPUT_ACCELEROMETER=0` in a udev rule? See:
https://github.com/systemd/systemd/blob/master/hwdb/60-sensor.hwdb
for details on how

janbrummer, 2019-09-02:
Thanks for your quick response Bastien! I've fixed it by adding another quirk for my system and created a pull request for systemd: https://github.com/systemd/systemd/pull/13453

Therefore this one is obsolete. Thanks.

closed 2019-09-02 by janbrummer