hadess/iio-sensor-proxy #285: Question not a bug.

srscott, 2019-09-26:
Which 0 or 1 does what?

sensor:modalias:acpi:SMO8500*:dmi:*:svnASUSTeK*:pnTP500LNG:*
 ACCEL_MOUNT_MATRIX=1, 1, 1; 1, 1, 1; 1, 1, 1

Like what does the first 1 do? the second?
normal, left, inverted, right?

I cannot find anything online to tell me which "switch" does what?

srscott, 2019-09-26:
The reason for asking is my screen is rotated 90 degrees to the right when i log into Arch Linux (Plasma) and i want to correct this and then make sure all other roatates function as they should.

hadess, 2019-10-08:
Pointing you to this is the best I can do right now:
https://docs.microsoft.com/en-gb/previous-versions/windows/apps/dn440593(v=win.10)

The matrix is a "3x3 matrix" which the accelerometer readings are multiplied by so for example:
accel reading: x = 0, y = -256, z = 0
matrix: 1, 0, 0; 0, 0, 1; 0, 1, 0
corrected reading: x = 0, y = 0, -256

I couldn't find better example on how to do this right now, sorry.

hadess, 2020-01-08:
Some notes for myself:

[KaTex](https://katex.org/) syntax for rotation matrix:
```
  \left[ {\begin{array}{ccc}
   x & y & z\\
  \end{array} } \right]
*
  \left[ {\begin{array}{ccc}
   x & y & z\\
   x & y & z\\
   x & y & z\\
  \end{array} } \right]
```

Gitlab KaTeX support:
https://gitlab.com/gitlab-org/gitlab/blob/master/doc/user/markdown.md#math

Some PDFs with explanations:
https://cache.freescale.com/files/sensors/doc/app_note/AN3461.pdf
https://www.degruyter.com/downloadpdf/j/acsc.2016.26.issue-2/acsc-2016-0013/acsc-2016-0013.pdf
http://www.irrd.ca/cag/smartphone/MEMEA2013-Smartphone-axis.pdf

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/285

closed 2020-01-09 by hadess