hadess/iio-sensor-proxy #287: Make iio-sensor-proxy only rotates for 2-in-1 or convertibles

khfeng, 2019-11-08:
Many laptops use accelerometer for shell shock protection instead of screen rotation, so screen rotating on a workstation laptop is an unexpected behavior.

We can either blacklist chassis type Portable (8), Laptop (9) and Notebook (10), or whitelist Tablet (30), Convertible (31) and Detachable (32) for this matter.

hadess, 2019-11-08:
> Many laptops use accelerometer for shell shock protection instead of screen rotation, so screen rotating on a workstation laptop is an unexpected behavior.

By "many laptops" I guess you mean your laptop. What laptop is it? If the accelerometer isn't in the display, then you should probably add the necessary quirk to systemd so iio-sensor-proxy knows it's in the base and ignores it:
https://github.com/systemd/systemd/blob/master/hwdb.d/60-sensor.hwdb#L60

> We can either blacklist chassis type Portable (8), Laptop (9) and Notebook (10), or whitelist Tablet (30), Convertible (31) and Detachable (32) for this matter.

That's not going to work because there are plenty of laptops and devices with the wrong info, or that straddle 2 types. Eg. a Lenovo Yoga could easily be marked with any of the first 3 chassis types you wanted to blacklist.

khfeng, 2019-11-09:
Many laptops, mostly HP :)
But apparently HP's lis3lv02d is covered systemd's new hwdb.

Now I know ACCEL_LOCATION=base to address what I intend to do, I'll send a PR there.

closed 2019-11-09 by khfeng

hadess, 2019-11-09:
> Many laptops, mostly HP :)

I guess there's good changes it's already fixed then.