hadess/iio-sensor-proxy #288: Can't find i2c-BOSC0200:00/iio:device0

giovannicaligaris, 2019-11-13:
I have a tablet by ALLDOCUBE iWork10 Pro i15-TD and the orientation of the screen is wrong. Currently when sitting on landscape it gives me bottom-up when it should be right-up

$ monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: bottom-up)
=== No ambient light sensor
    Accelerometer orientation changed: bottom-up
    Accelerometer orientation changed: right-up
    Accelerometer orientation changed: bottom-up

Also, the systemd service has a hard time finding the sensor.

iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: enabled)
   Active: active (running) since Sat 2019-11-09 12:51:33 GMT; 18min ago
 Main PID: 4182 (iio-sensor-prox)
    Tasks: 3 (limit: 4566)
   CGroup: /system.slice/iio-sensor-proxy.service
           └─4182 /usr/sbin/iio-sensor-proxy

Nov 09 12:51:33 tablet-i1002SG systemd[1]: Starting IIO Sensor Proxy service...
Nov 09 12:51:33 tablet-i1002SG systemd[1]: Started IIO Sensor Proxy service.
Nov 09 12:51:33 tablet-i1002SG iio-sensor-prox[4182]: Could not find trigger name associated with /sys/devices/pci0000:00/808622C1:02/i2c-2/i2c-BOSC0200:00/iio:device0

I have created file /lib/udev/hwdb.d/61-sensor-local.hwdb

sensor:modalias:acpi:BOSC0200*:dmi*
 ACCEL_MOUNT_MATRIX=1, 0, 0; 0, 1, 0; 0, 0, 1

And have updated after

sudo systemd-hwdb update
sudo udevadm trigger

After running udevadm info -export-db | grep ACCEL it will not find anything.

Help please.

I also posted this on launchpad https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1851935

hadess, 2020-01-08:
What's the output of `tree /sys/devices/pci0000:00/808622C1:02/i2c-2/i2c-BOSC0200:00/iio:device0` ?

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/288

closed 2020-01-09 by hadess