hadess/iio-sensor-proxy #290: Acer C720P

hitech95, 2019-12-20:
I'm doing something wrong for sure.
No idea why it fails to find the sensor.

I can probe it by sysfs without any problem.

Any idea on how to debug this?

I'm also having problems building from source (it says GTK not found but its installed)
For this reason I'm using the 2.4 package provided by the Ubuntu repository.

```
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:3219): DEBUG: 16:14:15.765: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ sudo !!
sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:3225): DEBUG: 16:14:25.161: Could not find any supported sensors        
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ cat in_illuminance0_input 
10
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ cat in_illuminance0_input 
332
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ udevadm info --export-db | grep iio
P: /devices/pci0000:00/INT33C3:00/i2c-1/1-0044/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/pci0000:00/INT33C3:00/i2c-1/1-0044/iio:device0
E: DEVTYPE=iio_device
E: SUBSYSTEM=iio
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
Errore: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files
```

hadess, 2020-01-08:
Looks like the code needs to handle `illuminance0` in addition to `illuminance` as possible channel names.

hitech95, 2020-01-08:
@hadess I'm having some issues on building. It ask me for GTK3 but its installed. I'm missing something?

hadess, 2020-01-08:
You need to install the development packages. If you don't manage to build the package, or test it, please ask the Ubuntu package maintainer to create a test package for you.

hitech95, 2020-01-08:
> 
> 
> You need to install the development packages. If you don't manage to build the package, or test it, please ask the Ubuntu package maintainer to create a test package for you.

CCould you provide the full list of depensencies needed for building it?

hadess, 2020-01-09:
> Could you provide the full list of dependencies needed for building it? Or publish how to build it?

It's built like every package that uses autotools, there's really nothing more to it.

The full list of depedencies expressed as Ubuntu packages are in the description (`.dsc`) file for the package:
https://packages.ubuntu.com/eoan/iio-sensor-proxy

So:
`libgudev-1.0-dev, udev, systemd, gtk-doc-tools (>= 1.0), libgtk-3-dev`

`cc1: all warnings being treated as errors`

Pass `--disable-Werror` to your `configure` call. I don't know why Ubuntu passes this flag, but it's certainly annoying.

hitech95, 2020-01-09:
Uhm... It still does not recognize the sensor.
As reference the sensor is an ISL29023.

```
nicolo@nicolo-Peppy:~/iio-sensor-proxy$ systemctl status iio-sensor-proxy.service
● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: enabled)
   Active: inactive (dead)

gen 09 12:28:07 nicolo-Peppy systemd[1]: Starting IIO Sensor Proxy service...
gen 09 12:28:07 nicolo-Peppy systemd[1]: Started IIO Sensor Proxy service.
nicolo@nicolo-Peppy:~/iio-sensor-proxy$ G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:2485): DEBUG: 12:29:00.720: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.
nicolo@nicolo-Peppy:~/iio-sensor-proxy$ sudo !!
sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:2490): DEBUG: 12:29:08.855: Could not find any supported sensors
nicolo@nicolo-Peppy:~/iio-sensor-proxy$ git status
Sul branch wip/hadess/illuminance0
Your branch is up to date with 'origin/wip/hadess/illuminance0'.

nicolo@nicolo-Peppy:~/iio-sensor-proxy$
```

hadess, 2020-01-09:
> ** (process:2485): DEBUG: 12:29:00.720: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.

You probably need to reboot. Attach the output of `udevadm info --export-db` after a reboot.

hitech95, 2020-01-09:
I've already trie dto reboot. I've tried again and now it works :D
No idea why it was stuck!

You can merge it into master 👍

hadess, 2020-01-09:
>  As reference the sensor is an ISL29023.

Huh, this sensor's driver (`isl29018`) doesn't use `illuminance0`, only the 2 drivers I mentioned do.

Can you please attach the output of `udevadm info --export-db` as I requested?

hitech95, 2020-01-09:
> > As reference the sensor is an ISL29023.
> 
> Huh, this sensor's driver (`isl29018`) doesn't use `illuminance0`, only the 2 drivers I mentioned do.
> 
> Can you please attach the output of `udevadm info --export-db` as I requested?

In my case it's using illuminance0.

```
nicolo@nicolo-Peppy:~$ cd /sys/bus/iio/devices/iio\:device0/
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ cat name 
i2c-ISL29023:00
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ ls -la
totale 0
drwxr-xr-x 3 root root    0 gen  9 15:50 .
drwxr-xr-x 4 root root    0 gen  9 15:50 ..
-r--r--r-- 1 root root 4096 gen  9 15:54 dev
-rw-r--r-- 1 root root 4096 gen  9 15:54 in_illuminance0_calibscale
-rw-r--r-- 1 root root 4096 gen  9 15:50 in_illuminance0_input
-rw-r--r-- 1 root root 4096 gen  9 15:54 in_illuminance0_integration_time
-rw-r--r-- 1 root root 4096 gen  9 15:54 in_illuminance0_scale
-r--r--r-- 1 root root 4096 gen  9 15:54 in_illuminance_integration_time_available
-r--r--r-- 1 root root 4096 gen  9 15:54 in_illuminance_scale_available
-rw-r--r-- 1 root root 4096 gen  9 15:54 in_intensity_ir_raw
-r--r--r-- 1 root root 4096 gen  9 15:54 name
drwxr-xr-x 2 root root    0 gen  9 15:54 power
lrwxrwxrwx 1 root root    0 gen  9 15:54 subsystem -> ../../../../../../bus/iio
-rw-r--r-- 1 root root 4096 gen  9 15:54 uevent
nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$ udevadm info --export-db
P: /devices/LNXSYSTM:00
E: DEVPATH=/devices/LNXSYSTM:00
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXSYSTM:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=4299367

P: /devices/LNXSYSTM:00/BOOT0000:00
E: DEVPATH=/devices/LNXSYSTM:00/BOOT0000:00
E: ID_VENDOR_FROM_DATABASE=Coreboot Project
E: MODALIAS=acpi:BOOT0000:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=4300650

P: /devices/LNXSYSTM:00/LNXCPU:00
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:00
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=4301842

P: /devices/LNXSYSTM:00/LNXCPU:01
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:01
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=4302965

P: /devices/LNXSYSTM:00/LNXPWRBN:00
E: DEVPATH=/devices/LNXSYSTM:00/LNXPWRBN:00
E: DRIVER=button
E: ID_VENDOR_FROM_DATABASE=The Linux Foundation
E: MODALIAS=acpi:LNXPWRBN:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=4304216

P: /devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
E: DEVPATH=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
E: EV=3
E: ID_FOR_SEAT=input-acpi-LNXPWRBN_00
E: ID_INPUT=1
[... 5053 lines cut ...]
N: vcsu6
E: DEVNAME=/dev/vcsu6
E: DEVPATH=/devices/virtual/vc/vcsu6
E: MAJOR=7
E: MINOR=70
E: SUBSYSTEM=vc

P: /devices/virtual/vtconsole/vtcon0
E: DEVPATH=/devices/virtual/vtconsole/vtcon0
E: SUBSYSTEM=vtconsole

P: /devices/virtual/vtconsole/vtcon1
E: DEVPATH=/devices/virtual/vtconsole/vtcon1
E: SUBSYSTEM=vtconsole

P: /devices/virtual/workqueue/blkcg_punt_bio
E: DEVPATH=/devices/virtual/workqueue/blkcg_punt_bio
E: SUBSYSTEM=workqueue

P: /devices/virtual/workqueue/writeback
E: DEVPATH=/devices/virtual/workqueue/writeback
E: SUBSYSTEM=workqueue

nicolo@nicolo-Peppy:/sys/bus/iio/devices/iio:device0$
```

hadess, 2020-01-09:
Alright, that'd because it has multiple channels.

In the kernel's `industrialio-core.c`:
```C
case IIO_SEPARATE:
        if (chan->indexed)
                name = kasprintf(GFP_KERNEL, "%s_%s%d_%s",
                                    iio_direction[chan->output], // "in"
                                    iio_chan_type_name_spec[chan->type], // "illuminance"
                                    chan->channel, // "0" 
                                    full_postfix); // "input"
```

commit 3b5855ea (2020-01-09): light: Add support for "illuminance0" light sensors

That would be used in the tsl2772 and lm3533 kernel drivers, as well as
devices with multiple channels, such as combined proximity and light
sensors, which use "illuminance0" as the channel name instead of
"illuminance".

Closes: #290
PR #295 (2020-01-09): light: Add support for tsl2772 and lm3533 kernel drivers
Those use the "illuminance0" channel name instead of "illuminance".

Closes: #290
commit 4dda433e (2020-01-09): light: Add support for "illuminance0" light sensors

That would be used in the tsl2772 and lm3533 kernel drivers, as well as
devices with multiple channels, such as combined proximity and light
sensors, which use "illuminance0" as the channel name instead of
"illuminance".

Closes: #290
closed 2020-01-09 by hadess