hadess/iio-sensor-proxy #296: Not working on Clear Linux

bedawi, 2020-02-22:
HI there,

I am trying to install the IIO Sensor Proxy on Clear Linux. I cloned the repo and proceeded like this:

```
./autogen.sh
make
sudo make installl
```

```udevadm info --export-db``` shows this amongst a lot more:

```
P: /devices/pci0000:00/0000:00:17.2/i2c_designware.6/i2c-11/i2c-KIOX000A:00/iio:device0
N: iio:device0
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:17.2/i2c_designware.6/i2c-11/i2c-KIOX000A:00/iio:device0
E: SUBSYSTEM=iio
E: DEVNAME=/dev/iio:device0
E: DEVTYPE=iio_device
E: MAJOR=239
E: MINOR=0
E: USEC_INITIALIZED=5552361
E: ACCEL_MOUNT_MATRIX=1, 0, 0; 0, -1, 0; 0, 0, 1
E: IIO_SENSOR_PROXY_TYPE=iio-buffer-accel
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
```

More output:

```$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
Error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files
```

```
$ sudo systemctl status iio-sensor-proxy.service
● iio-sensor-proxy.service - IIO Sensor Proxy service
     Loaded: loaded (/usr/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: disabled)
     Active: inactive (dead) since Sat 2020-02-22 17:38:24 CET; 7s ago
    Process: 3880 ExecStart=/usr/sbin/iio-sensor-proxy (code=exited, status=0/SUCCESS)
   Main PID: 3880 (code=exited, status=0/SUCCESS)

Feb 22 17:38:24 s11b systemd[1]: Starting IIO Sensor Proxy service...
Feb 22 17:38:24 s11b systemd[1]: iio-sensor-proxy.service: Succeeded.
Feb 22 17:38:24 s11b systemd[1]: Started IIO Sensor Proxy service.
```

I guess this could have something to do with Clear Linux. Unfortunately journalctl does not provide a clue whats wrong. Here is the journal's output when restarting the iio-sensor-proxy.service:

```
Feb 22 17:30:59 s11b systemd[1]: Starting IIO Sensor Proxy service...
Feb 22 17:30:59 s11b systemd[1]: iio-sensor-proxy.service: Succeeded.
Feb 22 17:30:59 s11b systemd[1]: Started IIO Sensor Proxy service.
```

I turned on debugging and got this output:
```
iio-sensor-prox[4109]: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.
```

...and no, the service is not already running.

I had also provided a custom a custom 61-sensor-local.hwdb-file in /etc/udev/hwdb.d/ as I describe on my GitHub repo here: https://github.com/bedawi/trekstor-s11b/tree/master/systemd - I do not think this is related to the problem but who knows?!

hadess, 2020-02-22:
> ```
> iio-sensor-prox[4109]: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.
> ```
> ...and no, the service is not already running.

Then it's the other thing.

You didn't install iio-sensor-proxy properly and D-Bus can't find the `.service` files associated with it. You probably want to pass `--prefix=/usr --sysconfdir=/etc`, or ask the ClearLinux distribution about this.

closed 2020-02-22 by hadess

bedawi, 2020-02-22:
> Then it's the other thing.
>
> You didn't install iio-sensor-proxy properly and D-Bus can't find the `.service` files associated with it. You probably want to pass `--prefix=/usr --sysconfdir=/etc`, or [...]

Thanks for your quick reply. Maybe a stupid question: Pass the parameters where? There is no ./configure file in the folder.

bedawi, 2020-02-22:
OK, just for follow up, so that others, who repeat my mistakes, find a solution here:

The command is: ```./autogen.sh --prefix=/usr --sysconfdir=/etc```

I had to uninstall the old files with ```sudo make uninstall``` and I had to clone a new copy of the repository. Then compiling went through and after a reboot the device worked.

@hadess: maybe you like to update your instructions on https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/blob/master/README.md - thanks for your work!